actuallymentor/battery #1: Any plans for adding this tool to Homebrew?

mihaiconstantin, 2022-04-04:
Thanks for this super handy tool! 
Have you considered adding it to Homebrew? 
Also, is there an uninstall script? Or at least the steps one would need to follow?

actuallymentor, 2022-09-06:
@mihaiconstantin  You have docs that describe how to submit tools with a setup scrip that requires superuser permissions?

mihaiconstantin, 2022-09-08:
@actuallymentor No, I don't have any info on that.

@JacksonChen666 I think your suggestion seems more reasonable.

Pyciko, 2022-09-18:
Just wanna say thank you fellow person for releasing this. AlDente with its 30$ lifetime (or 15$ per year) pro version just to hide the ****ing menubar icon is insane...

actuallymentor, 2022-10-13:
@mihaiconstantin you can now install this as a cask: `brew install battery`. It installs the GUI app which in the background installs the CLI.

Please let me know if it works for you, I have no fresh device to test it on.

Ps. @pyciko you're welcome

closed 2022-10-13 by actuallymentor

pyciko, 2022-10-13:
worked like charm!

mihaiconstantin, 2022-10-13:
@actuallymentor, it works brilliantly! Awesome job!