actuallymentor/battery #4: Not working at macOS 12.3.1(M1 Pro MacBookPro 14")

Steve-Xyh, 2022-05-02:
Thanks for your scripts!
But after I run `battery charging off` the battery icon does not  have any changes...😢

Steve-Xyh, 2022-05-02:
I checked the `CH0B` and `CH0C` values, they are set to `00` after running `battery charging on`, which makes sense.
But they are set to `03` after running `battery charging off`, it should be `02`. I don't know about SMC hardware things, so I'm not sure what happened. I would be very grateful if you could help me.

This is the script output
[image]

anarchy89, 2022-08-02:
I had this problem on my MacBook 16 m1 max, I typed battery charging off, then I removed the power cable and plugged it back in and now it works as intended.

kitrak4002, 2022-08-20:
Hello, I'm a new Macbook M1 user and not into as much coding and I don't have much experience using the terminal. I used the one-line installation for the Battery cli tool. However, I don't know how to use the commands in the terminal for the Battery cli tool. It would be helpful if anyone posted specifically what to type to get the commands going. Thank you!

tob-00, 2022-08-23:
> Hello, I'm a new Macbook M1 user and not into as much coding and I don't have much experience using the terminal. I used the one-line installation for the Battery cli tool. However, I don't know how [...]

Hi! it worked for me running it as administrator (MacBook Air M2): 
`sudo curl https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | sudo bash`

Hope that helps :)

kitrak4002, 2022-08-27:
Thank you so much! It is working now!

anarchy89, 2022-08-31:
> Hi! it worked for me running it as administrator (MacBook Air M2): `sudo curl https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | sudo bash`
>
> Hope that helps :)

Hi @tob-00 do you have to unplug and plug back in the charger whenever you do battery charging off in the terminal?

tob-00, 2022-08-31:
> Hi @tob-00 do you have to unplug and plug back in the charger whenever you do battery charging off in the terminal?

No I don't but it takes some time. I think after like every percentage charged it is evaluated if charging should be continued and then it is turned off...

actuallymentor, 2022-09-06:
I edited the setup script. It should now work by running:

```
curl -s https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | bash
```

closed 2022-09-06 by actuallymentor