actuallymentor/battery #5: Starting at login

tob-00, 2022-08-28:
Hello :) I added this code to the beginning of the charging routine in order to prevent the program to run if the Mac is not plugged into power:
`power_plugged_in=pmset -g batt | tail -n1 | awk {print $4}`
`if [[ "$power_plugged_in" == "discharging;" ]]; then`
    `exit 0`
`fi`
If I run it from the terminal, it works perfectly fine (it just stops if not plugged into power). However, I am struggling to add it as a log in item using automator. I tried `usr/local/bin/battery charge 80` and I used "as arguments" and added it as a log in item. However, after the startup it keeps charging, even though testing in automator shows the correct result. Any ideas on that? Thanks a lot in advance!!

tob-00, 2022-08-30:
I found it out myself. I used bin/bash and "as arguments" together with this code:
> #!/usr/bin/env bash
> export PATH=/usr/local/bin:$PATH
> battery "charge" "80"

actuallymentor, 2022-09-06:
Hey!

Little update for you: you can use the command `battery maintain 80` to maintain a battery level. I'd recommend just running:

```
nohup battery maintain 80&
```

Which will make it run in the background