actuallymentor/battery #8: Error when running battery charging off with latest version

anarchy89, 2022-09-26:
After updating the app, when I run `battery charging off`, I keep getting the following error,

`
touch: /Users/xxx/.battery/battery.log: Permission denied
15:47:15 - Setting charging to off
touch: /Users/xxx/.battery/battery.log: Permission denied
touch: /Users/xxx/.battery/battery.log: Permission denied
15:47:15 - Enabling battery charging
touch: /Users/xxx/.battery/battery.log: Permission denied
15:47:15 - Battery at 63% ((no remaining), smc charging enabled
15:47:15 - Disabling battery charging
`

I am unable to access the .battery folder. I have tried deleting the folder and reinstalling battery as well. Still get the same error.

I also ran sudo battery visudo. Still have the same error.

actuallymentor, 2022-09-27:
You'll have to run `sudo rm -rf ~/.battery` and `battery reinstall` in theory it should work after. 

Please let me know if that fixes it.

anarchy89, 2022-09-27:
Hey it worked @actuallymentor , thanks! You should add this to the readme. I reckon it happens a lot.

Evertt, 2022-10-08:
I have the same problem.

> You'll have to run sudo rm -rf ~/.battery and battery reinstall in theory it should work after.

This did not fix it for me. And just want to add one more interesting bit of information. When I run `battery` in my terminal, the first line of output says `touch: /Users/evert/.battery/battery.log: Permission denied`.

Then as a sanity check, I just tried to run that command myself: `touch /Users/evert/.battery/battery.log` and even _I_ got `Permission denied`. Which I thought was odd, because I assumed that I could do anything I pleased in my own home folder.

After running `ls -al ~/.battery` I see that `battery.log` actually already exists and that is owned by the root user.

Even though I'm sure that nowhere during the installation process did I intentionally give any process root acces. But it may of course be possible, because the app did ask me to fill in my password one time to install additional components. So maybe the app is using that password accidentally to give itself root access while installing those additional components. And so yes, if it then also created `/Users/evert/.battery/battery.log` then that could explain why that file now belongs to the root user.

So I just deleted that log file by force. And then when I ran `battery` again, it didn't show an error as the first line. Then when I ran `ls -al ~/.battery` I could see a newly created `battery.log` file which is not owned by root anymore, but instead by me. And now when I open up the app, I don't get those error popups that OP showed.

Hope this information can help you to fix this issue better.

chrisfromredfin, 2022-10-10:
I installed with homebrew and was having this issue. I just ran `sudo chown $(whoami) /Users/$(whoami)/.battery/battery.log` and it stopped.

actuallymentor, 2022-10-13:
Hey all!

Thanks for doing a dive into this, I appreciate it.

This issue should only occur if you had a previous version installed. The beta/RC versions messed up some things (my bad), and it should not happen in the future.

If anyone finds this thread, things you can do to attempt a fix:

```
sudo rm -rf ~/.battery
binfolder=/usr/local/bin
sudo rm -v "$binfolder/smc" "$binfolder/battery"
```

And to reinstall use brew (if you have it):

```
brew install battery
```

closed 2022-10-13 by actuallymentor

wesnuhin, 2025-07-04:
[image] is there any way to fix it?