actuallymentor/battery #9: black / white icon depending on status bar color

sataset, 2022-09-28:
Icon is adapted for light status bar only.

[image]

These threads might be helpful:
- [link 1](https://stackoverflow.com/questions/62685948/macos-big-sur-detect-dark-menu-bar-system-tray%29)
- [link 2](https://developer.apple.com/forums/thread/662322)

actuallymentor, 2022-10-13:
Ah weird, it should work:

[image]

What version number does `battery` show?

sataset, 2022-10-13:
Installed the latest 1.0.3 from brew and experienced the same effect.

I think the icon becomes normal during the charge or wallpaper change, but clicking "Disable 80% battery limit" reproduces the issue.

Currently this happens for me on both light and dark macOS themes with Catalina dynamic wallpaper during the night time.

waterlou, 2022-10-15:
same here, 1.0.3, installed from homebrew.

actuallymentor, 2022-12-17:
Note to self: might look at at some point. No high prio. If anyone figures out the source of this I'd welcome a PR.

actuallymentor, 2023-02-05:
Closing as I can't reproduce.

closed 2023-02-05 by actuallymentor