actuallymentor/battery #13: Feature request: low threshold to resume charging

chrisfromredfin, 2022-10-10:
It seems like the "maintain" command will cycle the charger in a pretty tight window. I wonder if it could easily enough be made configurable about when to turn the charging on to create a wider cycle. Is that even something that should happen?

```bash
if [[ "$battery_percentage" -gt "$setting" && "$is_charging" == "enabled" ]]; then

			log "Charge above $setting"
			disable_charging

		elif [[ "$battery_percentage" -lt "$setting" && "$is_charging" == "disabled" ]]; then

			log "Charge below $setting"
			enable_charging

		fi
```

I'm thinking something like `if -gt $setting disable_charging`, but `elif -lt $low_setting enable_charging`

Is this something you'd consider? I was told it's good to let the battery drain down by an Apple tech, so occasionally I'd like to set it to go down to 5% then recharge to 80%. 

Maybe make the second argument optional, like
`battery maintain 80` does the original behavior, but `battery maintain 80 10` does the new?

actuallymentor, 2022-10-13:
Do I understand correctly you want to have a type of "cycle battery" function that discharges to a low level and then recharges?

That used to be the case with old batteries, but Li-ion ones don't benefit from that, see [this article on batteries](https://batteryuniversity.com/article/bu-808-how-to-prolong-lithium-based-batteries):

> There is no memory and the battery does not need periodic full discharge cycles to prolong life

chrisfromredfin, 2022-10-13:
That's true that is what I'm asking for. I had heard from the guy at the Apple store when I brought it in for repair to discharge down sometimes, and not leave it plugged in all the time at 100. However, I suppose you're saying keeping it at 80% instead would solve the same problem? I had always understood the same about Li-Ion as you say, but Apple Store Guy™ got me second-guessing.

actuallymentor, 2022-10-13:
See the above article, in essence: keep Li-ion from going low and high and it will live longer. Bonus points for not letting it get hot.

There is to my knowledge no reason to periodically discharge.

closed 2022-10-13 by actuallymentor