actuallymentor/battery #15: "Error installing battery limiter: undefined" when launching UI on Monterey 12.5.1

lekv, 2022-10-16:
I installed Battery v1.0.3 using homebrew and then started the UI. After installing the command line component it shows a dialog saying `Battery limiter error: undefined` and after relaunching the UI, it shows a dialog with the error message `Error installing battery limiter: undefined`. Not sure what to try to fix this, let me know if you need any other information.

lekv, 2022-10-16:
Huh, this made the error go away: `sudo chmod a+w /Users/$(whoami)/.battery/battery.log`

Looks like a permission issue with this file, which after installation had `-rw-r--r--  1 root  staff` permissions

actuallymentor, 2022-10-18:
Well done on the fix! Did you have another version (the CLI only?) of `battery` installed prior to installing with Homebrew?

lekv, 2022-10-18:
I hadn't, and I repro'd it after deleting the `~/.battery` folder.

actuallymentor, 2022-10-19:
Thanks for the details, I'll add this to the list :)

closed 2022-12-12 by actuallymentor

nhooey, 2023-04-10:
I get the error regardless of the permissions of anything in `~/.battery`, and it only happens when launching Battery from Spotlight when it's not already running.

**It happens in these exact circumstances:**

Given:
1. Battery installed only ever from Homebrew cask, updated today
2. Nothing has been installed outside of Homebrew, Nix, and the Mac App Store
3. Run "Battery" from Spotlight, when the Battery application is not running

This error shows:
[image]

If I run `open /Applications/Battery.app`, the Battery app starts and shows its menulet in the Notification Area:

[image]

Then further invocations of "Battery" from Spotlight have no effect (probably because Mac OS doesn't actually do anything after a Spotify invocation of an app it detects is already running).

If I kill the Battery app by clicking "Quit" on its menulet dropdown menu, then all invocations of "Battery" from Spotlight show the error dialogue shown above in this message, until battery is run from `open /Applications/Battery.app` again.

So there is something wrong specifically with the Spotify invocation of Battery.

Running `sudo chmod -Rc a+w ~/.battery` has no effect in terms of preventing the error message.