actuallymentor/battery #16: Error installing battery limiter: Command failed: ./applet /bin/sh: ./applet: Bad CPU type in executable

jxnding, 2022-10-17:
After installing, on first-time launch, I get 2 errors:

Error installing battery limiter: Command failed: ./applet
/bin/sh: ./applet: Bad CPU type in executable

Battery limiter error: Command failed: PATH=$PATH:/usr/local/bin/battery battery status
/bin/bash: battery: command not found

I'm on Ventura 13.0 beta, installed the dmg battery-1.0.3-arm64.dmg

jxnding, 2022-10-17:
Same with brew install battery. However the CLI works kinda (I suspect because it compiles from source right?)

However, it's not working very well. Typing "battery status" gives:
touch: /Users/ding/.battery/battery.log: Permission denied

jxnding, 2022-10-19:
No, I am on an M1 Macbook Air.

actuallymentor, 2022-10-19:
That is very strange, since the error seems to think your processor is not...

What is your model when you click "about mac"?

[image]

timcampbell092, 2022-10-19:
Hi guys - same issue here on my M1 Macbook Air (Monterey 12.6). 
[image]
Tried the install two days ago and getting the same errors. Have tried installing via DMG file and via the command line in terminal. 
Getting the exact same errors.
[image]
[image]
New to MacOS on a new Macbook Air.
The MacBook currently has the following apps installed:
DisplayLink Manager
MS Office
Adobe Premiere & Photoshop Elements 2021
Magnet
MS Edge (Still learning Safari!)

jxnding, 2022-10-20:
[image]

jxnding, 2022-10-23:
Cloning the repo, checking out the latest version tag, and building does seem to work. Will report later (at least adds "battery" as a background item) 
[image]

merlinmah, 2022-11-26:
I get the same errors when Installing through `brew` in Ventura 13.0.1 on an M1 Max; I chalked it up to `/usr/local/bin` being hardcoded somewhere, instead of the `/opt/homebrew` prefix that `brew` uses by default. Installing via the Github shell script controls charging as expected, provided one remembers to use `sudo`. Thanks for making this!

actuallymentor, 2022-12-12:
@merlinmah how would that path cause a CPU error?

Tokarak, 2022-12-12:
It's not a cpu error, its a path error. Github search reveals all the hardcoded `/usr/local/bin/battery`.
https://github.com/actuallymentor/battery/search?q=%2Fusr%2Flocal%2Fbin%2Fbattery

actuallymentor, 2022-12-17:
@Tokarak thanks for pointing that out. I think I fixed that path issue just now. Can you confirm it was fixed?

phewitt, 2022-12-22:
looks like it fixed it for me. however the brew package and dmg still need a to include it. I had to clone the repo and run `./setup.sh` then install the GUI

chujiluo, 2022-12-29:
facing the same error on M2. Is this expected?

whiterock, 2022-12-30:
yeah the brew formulae need updating.

actuallymentor, 2023-01-05:
@phewitt that is weird because the brew `dmg` clones the repo using the latest version...

K-J-E, 2023-01-20:
hi, I am having this same error (installed through dmg on release page) and the command line commands do not work, so I am unable to use 'battery uninstall'. How could I uninstall the tool? thanks

actuallymentor, 2023-02-02:
@K-J-E you can manually delete the files

actuallymentor, 2023-02-04:
I just pushed a update that should fix a few issues, could you try to run `brew reinstall --cask battery` and see if that fixes it?

actuallymentor, 2023-02-05:
Could you try the following things to help me debug:

**Debug trail 1: installing CLI only**

Instead of installing through brew, can you install through [these CLI instructions](https://github.com/actuallymentor/battery#installation-1) and show me your terminal output?

**Debug trail 2: running the app locally**

Can you try:

```shell
git clone git@github.com:actuallymentor/battery.git
cd battery/app
npm i
npm start
```

And show me what happens?

KarstenS123, 2023-02-10:
I think my issue is the same in this issue report. If not, I can open a new one.

I tried just now to install the app by using the _battery-1.0.7-arm64.dmg_ file downloaded from the releases page.

My Mac is:

[image]

After enter the password for the "Updates" I get these error messages in this order:

[image]

[image]

[image]

[image]

The battery icon is in the menu bar, but nothing happens when I click on it.

Fiskheimer, 2023-02-11:
> I think my issue is the same in this issue report. If not, I can open a new one.
> 
> I tried just right now to install the app by using the _battery-1.0.7-arm64.dmg_ file downloaded from the releases page.
> 
> My Mac is:
> 
> [image]
> 
> After enter the password for the "Updates" I get these error messages in this order:
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image]
> 
> The battery icon is in the menu bar, but nothing happens when I click on it.

I get the same error messages and behavior on my MacBook Air M2.

[image]

markspercyspencer, 2023-02-12:
I have the exact same error messages when trying to install Battery.dmg on my M1 MacBook Air. 
[image]
[image]
[image]
[image]
[image]

actuallymentor, 2023-02-12:
I believe we found the issue in #48!

Can someone (@jxnding?) with this issue please install [this v1.0.8 pre-release](https://github.com/actuallymentor/battery/releases/tag/v1.0.8-beta) and let me known if that solves the issue?

closed 2023-02-13 by actuallymentor

aresYio, 2023-02-13:
It doesn't work, even with the 1.08 version

michkz, 2024-03-22:
I am also experiencing the same error: "error installing battery limited: undefined” after installing through brew. Im currently on the latest release (v1.2.1). The CLI installation does successfully install the application, but the commands “battery maintain 80” and “battery charging off” do not seem to show the expected visual effect on the battery icon. Also running on a M1 mac