actuallymentor/battery #20: Looking for a discharge functionality.

alaneuler, 2022-10-30:
Hi, great work here!

I have just one little additional problem. New monitor with type-C port have typically charging capability, which means you can  use an external monitor while charging in a single type-C cable. But if mac's battery exceeds the limit, I can't have it reduce to a lower percentage. Aldente has a **Discharge** functionality which is exactly what I'm describing (but I'm a CLI guy sadly).

Can you please work on this?

I'm willing to code and just don't know how to achieve this.

DUOLabs333, 2022-11-17:
Shouldn't it just be `battery charging off`?

shenlong2210, 2022-11-18:
If we limit changer at 80%. Then after 30 60 mins later, the current battery is 79% and the macbook will charge then. 
So can you made a script to limit chager 80% and when the current battery level 70% they have rechange again.

ps: AIDenta has this function and maybe the software alway read battery level informations so they make macbook drain enegry so much, maybe lost 10 20% overnight when close lid.

alaneuler, 2022-11-23:
> If we limit changer at 80%. Then after 30 60 mins later, the current battery is 79% and the macbook will charge then.
> So can you made a script to limit chager 80% and when the current battery level 70% they have rechange again.

But say if current SOC is 90% and the charge limit is set to 80%? There will be a long time the SOC of battery exceeds 80%.

> ps: AIDenta has this function and maybe the software alway read battery level informations so they make macbook drain enegry so much, maybe lost 10 20% overnight when close lid.

If I activates AlDente's "discharge' functionality, the macOS runs totally on battery (I can confirm this on menu bar's battery status) but not what you have said.

[image]

DUOLabs333, 2022-11-24:
I've been trying to use IOKit to limit charging, but it doesn't work. Looking at AlDente's issue tracker shows many people having problems with the Discharge functionality, so it's possible it doesn't work well at all.

alaneuler, 2022-11-25:
@DUOLabs333 Yes, absolutely. Even Aldente free version has this functionality, but it seems that code of "discharge" is not open sourced. You can find out [here](https://github.com/davidwernhart/AlDente-Charge-Limiter) by yourself.

DUOLabs333, 2022-11-25:
Yeah, I checked there, but didn't find it (it seems that the version on GitHub is not the same as the one put in the releases).

alaneuler, 2022-11-25:
Apple's open API doesn't provide what you want (besides IOKit, you can also check IOPMLib).

Aldente works well on Monterey but not for Ventura, it's not surprising though because most of the macOS battery related apps rely on the private API SMCKit.

alaneuler, 2022-11-25:
> Yeah, I checked there, but didn't find it (it seems that the version on GitHub is not the same as the one put in the releases).

sadly...

DUOLabs333, 2022-11-28:
The only I can think of that works is basically run a while loop to drain the battery, but that sounds incredibly wasteful and not guaranteed to work in all situations.

actuallymentor, 2022-12-12:
Hey @alaneuler did you find the `smc` code for disabling the adapter input? I don't have time to dig into the aldente source code but if you find the code I can add it over the xmas holiday.

alaneuler, 2022-12-17:
@actuallymentor it seems that Aldente doesn't open source its discharge functionality code.

actuallymentor, 2022-12-17:
Ah that's too bad. I'll leave this issue open.

If anyone figured out the right SMC key to disable adapter input (while the adapter is plugged in) please let me know.

DUOLabs333, 2022-12-19:
I found this: https://developer.apple.com/documentation/kernel/iopmpowersource/1809904-powersource, which has setIsCharging, but I can't make heads or tails of how it works (I can't even compile a test program with the appropriate header `Kernel/IOKit/pwr_mgt/IOPMPowerSource.`)

Howie65, 2022-12-24:
> Ah that's too bad. I'll leave this issue open.
> 
> If anyone figured out the right SMC key to disable adapter input (while the adapter is plugged in) please let me know.

You had a discussion with "aykevl" in Oct. (https://github.com/zackelia/bclm/issues/20). There was also a link to the source "macsmc_power.c" with the wanted information inside.

I did test the following and it seems to work:
sudo smc -k CH0I -w 01      // battery starts to discharge even with PS connected
sudo smc -k CH0I -w 00      // back to normal behavior

I also found the key "SBAS" (smc -k SBAS -r) which seems to deliver a more precise charge-level than "pmset -g batt". Guess AlDente uses SBAS.

DUOLabs333, 2022-12-24:
I just tried this in my own tool and it works perfectly.

DUOLabs333, 2022-12-24:
Though SBAS seems a minute behind pmset

Howie65, 2022-12-24:
> Though SBAS seems a minute behind pmset.

rather different values: https://apphousekitchen.com/feature-explanation-hardware-battery-percentage/

sofmz, 2023-01-03:
"sudo smc -k CH0I -w 01" works for me, it does make my m1 macbook to use battery to discharge, but it also invalidates clamshell mode.

howie65, 2023-01-05:
> ...it also invalidates clamshell mode.

what does it mean excactly? I don't have an external display to test

actuallymentor, 2023-01-05:
@howie65 thanks for the sleuthing! I'll add it to the todo list (a PR is welcome of course).

sofmz, 2023-01-05:
> > ...it also invalidates clamshell mode.
> 
> what does it mean excactly? I don't have an external display to test

If you execute the command while in clamshell mode, the clamshell mode will be deactivated and macbook goes to sleep immediately. The effect is exactly as you unplug power cable forcing macbook running in battery mode while in clamshell mode. 

Hope this is clear enough. Probably we need something to inhibit the sleep to make this works for clamshell mode.

howie65, 2023-01-05:
> Hope this is clear enough. Probably we need something to inhibit the sleep to make this works for clamshell mode.

Ok I understand. caffeinate command should help here.

actuallymentor, 2023-01-05:
@howie65 sounds like the easiest way to do this is to caffeinate [this line](https://github.com/actuallymentor/battery/blob/main/battery.sh#L278).

The thing I want to avoid is forcing the device to be awake when on battery power. That is a source of battery drain (Windows devices have this issue when put in a backpack for example).

Ideally we'd do:

```
if <check if adapter is connected>
  caffeinate sleep
else
  just sleep
```

Though I'm not sure what is the most elegant way to check for plugged-in status.

st-ivan, 2023-01-19:
> @howie65 sounds like the easiest way to do this is to caffeinate [this line](https://github.com/actuallymentor/battery/blob/main/battery.sh#L278).
>
> The thing I want to avoid is forcing the device to [...]

Hi, im willing to try this. Can you elaborate how to run the commands in order to achieve this? Im trying to:
1. Use clamshell mode always
2. Have the power adapter plugged in always.
3. Keep the battery charge at 80% always. 
4. Do a manual calibration every 2 weeks (for this im going to let the battery run down to 15%, then charge it up to 100%, stay there for 1 hour and then discharge it to 80%) 

Yeah i know aldente is supposed to do all this automatically, however, it does not. Not even with pro subscription. I just requested a refund after trying everything.

howie65, 2023-01-19:
> Hi, im willing to try this. Can you elaborate how to run the commands in order to achieve this?

Normally there should be all tools in the box now to go manually.
1. Install CLI Battery in the current version
2. If your battery is above 80% run "sudo smc -k CH0I -w 01" and "caffeinate" in the terminal
3. If your battery is equal or below 80% then stop caffeinate  and run "-sudo smc -k CH0I -w 00" in the terminal

This should be the steps but I cannot test because of missing external monitor. Please report if this works.

st-ivan, 2023-01-20:
thanks for you quick response. 
running cmd "sudo smc -k CH0I -w 01" puts the computer to sleep. 
So i ran it after running "caffeinate" and it still puts the computer to sleep. 
Meaning as soon as the computer detects the power adpater is "unplugged" in clamshell mode, it goes to sleep (normal macos behavior).

howie65, 2023-01-20:
Sorry to hear that caffeinate isn't the solution to prevent from sleep.

Google has led me to this page: https://beebom.com/use-macbook-clamshell-mode-without-power-adapter/
Maybe you like to give "Amphetamine" a chance?

st-ivan, 2023-01-21:
> Sorry to hear that caffeinate isn't the solution to prevent from sleep.
> 
> Google has led me to this page: https://beebom.com/use-macbook-clamshell-mode-without-power-adapter/ Maybe you like to give "Amphetamine" a chance?

hi, thanks for your help. I remember i tried this before and didnt work. but i was doing the procedure wrong.So after following all these steps, in order for it to work:
1. Open lid (get out for clamshell mode).
2. Start close lid session
3. disconnect power cable.
4. Enter clamshell mode (close lid).

Note: it will not work if you start the session (step 1) and disconnect the power. It will automatically go to sleep (normal macos behaviour). 
So i think with battery app at 80% + amphetamine I can manually calibrate and now keep my battery healthy.
Thanks all for your help.

actuallymentor, 2023-02-05:
Functionality for discharge is not added. See new help message of `battery`.

closed 2023-02-05 by actuallymentor