actuallymentor/battery #21: The tool stopped working in MacOS Ventura

cryptolife2022, 2022-10-30:
I tried installing it from the source [the standard installation (brew and DMG) is broken, doesn't install SMC (typo in install command line), and permission problems in battery.log file]

disabled charging (battery charging off) - Nothing happens even though the logs say it worked.

sataset, 2022-11-02:
I was able to use app, however it seems to be a bit inconsistent (I think battery got charged several percents during sleep)

- uninstall app and remove traces
- uninstall cli tool (`battery uninstall`)
- installing from brew
- start app
- go through init
- quit app
- disable app autostart
- `sudo battery maintain 80`

P.S. at the 2nd step I have received 2 messages in shell `sudo: smc: command not found`.

Rommelkontje, 2022-11-07:
The app works via the installation of brew, however when I restart the Mac, the settings are completely restored.
Then if I Re-enter code ' sudo battery maintain 80 '. Battery percentage is limited again.

Is there a way to fix this?

ballo, 2022-11-15:
Does it actually work in Monterey? I've never seen it work. Ever.

ballo, 2022-11-16:
> I was able to use app, however it seems to be a bit inconsistent (I think battery got charged several percents during sleep)
> 
> * uninstall app and remove traces
> * `battery uninstall`
> * install from brew
> * start app
> * go through init
> * quit app
> * disable app autostart
> * `sudo battery maintain 80`
> 
> P.S. at the 2nd step I have received 2 messages in shell `sudo: smc: command not found`.

I followed these instructions more or less and finally got it working in Monterey. I only used the GUI for the initial setup (whatever voodoo it does) and n0ow ONLY use the command line and it seems to work.

actuallymentor, 2022-12-12:
There were some bugs in the script. It should work fine now, if something does not, please reopen this issue and tell me what you are having issues with :)

actuallymentor, 2022-12-12:
> I only used the GUI for the initial setup (whatever voodoo it does)
The voodoo is that it runs the setup script (same as CLI instructions in `README.md`) but the `PATH` of the GUI app is a bit messed up so that caused issues.

ballo, 2022-12-20:
For whatever reason it started charging again until I reset it. I didn't reboot the computer. Anyone know what voodoo is resetting the setting to 100% charging?