actuallymentor/battery #25: "It's not working"

data-science-made-easy, 2022-11-10:
After 'brew install battery' and 'open /Applications/Battery.app', I get these two errors:

- [image]
- [image]

Whatever I configure, my M1 stays charging :(
[image]

The command line program gives no error:
[image]

Any help? :)

Rommelkontje, 2022-11-11:
I noticed that mine did stop before charging to a full 100% so that's good. But still it won't stop charging when it reaches the percentage I want it to be maintained to, but rather somewhat more.

mcastaldelli, 2022-11-13:
I'm getting the same error on an m2 macbook air running Ventura. Charging also continues despite the limit :-(

ballo, 2022-11-13:
My battery is back at 100% Is there some secret to get this to work?

data-science-made-easy, 2022-12-13:
Thanks for looking at this, Mentor. Unfortunately, I seem to be stuck in exactly the same situation as described above. I took a look at the three lines you have edited yesterday in battery.sh (you've added path + some administration). Can  you explain why you think that this should have solved the issue?

I figured out your script uses 'smc' under the hood. The two commands `smc -k CH0B -w 02` and `smc -k CH0C -w 02` should do the trick and stop the charging, if I understand battery.sh correctly. Running these two commands however don't seem to do anything :(

Can you explain why you think the exact above two commands should stop the charging? Any idea what can be wrong?

Thanks!