actuallymentor/battery #27: Battery limiter error any help plz?

Sandeep-Khemani1, 2022-11-17:
1. Error installing battery limiter: undefined
2. Battery limiter error: undefined

anshibanov, 2022-11-20:
+1 macbook air m1 macos big sur

swissbuechi, 2022-11-22:
+1 macbook pro 14 m1 pro ventura 13.0.1

blackketter, 2022-12-04:
+1 MacBook Air M2 Ventura

jpedigo, 2022-12-06:
+1 14" MacBook Pro M1 Pro 12.6.1 Monterey

jeyraof, 2022-12-08:
+1 13 M1 Air Ventura

Sven030, 2022-12-09:
+1 macbook pro 1 m1 pro ventura 13.0.1

actuallymentor, 2022-12-12:
Jesus I should really enable more Github notifications. Sorry about letting this get out of hand everyone.

A pull request form a kind stranger fixed a typo I made that caused this.

Everything should work now.

closed 2022-12-12 by actuallymentor

helicos6, 2022-12-12:
@actuallymentor How to download the fixed version? the DMG still says Oct 5

actuallymentor, 2022-12-12:
> @actuallymentor How to download the fixed version? the DMG still says Oct 5

The GUI pulls the latest CLI. So just quit and restart the app :)

actuallymentor, 2022-12-13:
I managed to reproduce the issue, try it now?

helicos6, 2022-12-13:
@actuallymentor Issue still persists.

actuallymentor, 2022-12-13:
Could you show me the content of ~/.battery/battery.log please?

helicos6, 2022-12-13:
@actuallymentor battery.log is empty.

gui.log data:

```

Starting tray app
Triggering boot-time auto-update
Tray app boot complete
Installing battery for praveen...
Is installed? 
false
Update process complete
Install result: 

####################################################################
# 👋 Welcome, this is the setup script for the battery CLI tool.
# Note: this script will ask for your password once or multiple times.
####################################################################

🔋 Starting battery installation

[ 1/9 ] Superuser permissions acquired.
[ 2/9 ] Cloning fan control version of smc
[ 3/9 ] Building smc from source
[ 4/9 ] Move smc to executable folder
[ 5/9 ] Cloning battery repository
[ 6/9 ] Writing script to /usr/local/bin/battery for user praveen
[ 7/9 ] Set up visudo declarations
[ 8/9 ] Removing temp folder /Users/praveen/.battery-tmp
[ 9/9 ] Removed temporary build files

🎉 Battery tool installed. Type "battery help" for instructions.

Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Starting tray app
Triggering boot-time auto-update
Tray app boot complete
Is installed? 
/usr/local/bin/battery

Updating battery...
Update process complete
Update/install error: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Enable limiter
Error enabling battery: 
touch: /Users/praveen/.battery/battery.log: Permission denied
/usr/local/bin/battery: line 299: /Users/praveen/.battery/battery.log: Permission denied
touch: /Users/praveen/.battery/battery.log: Permission denied

Tray app boot complete
Starting tray app
Triggering boot-time auto-update
Is installed? 
/usr/local/bin/battery

Updating battery...
Update process complete
Update/install error: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Triggering boot-time auto-update
Tray app boot complete
Starting tray app
Is installed? 
/usr/local/bin/battery

Updating battery...
Update/install error: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Update process complete
Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Starting tray app
Triggering boot-time auto-update
Tray app boot complete
Is installed? 
/usr/local/bin/battery

Updating battery...
Update process complete
Update/install error: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Tray app boot complete
Triggering boot-time auto-update
Starting tray app
Is installed? 
/usr/local/bin/battery

Updating battery...
Update/install error: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Update process complete
Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Starting tray app
Triggering boot-time auto-update
Tray app boot complete
Is installed? 
/usr/local/bin/battery

Updating battery...
Update process complete
Update/install error: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Tray app boot complete
Triggering boot-time auto-update
Starting tray app
Is installed? 
/usr/local/bin/battery

Updating battery...
Update process complete
Update/install error: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Starting tray app
Triggering boot-time auto-update
Tray app boot complete
Is installed? 
/usr/local/bin/battery

Updating battery...
Update process complete
Update/install error: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Starting tray app
Tray app boot complete
Triggering boot-time auto-update
Updating battery...
Is installed? 
/usr/local/bin/battery

Update process complete
Update/install error: 
touch: /Users/praveen/.battery/battery.log: Permission denied

Error getting battery status: 
touch: /Users/praveen/.battery/battery.log: Permission denied

```

helicos6, 2022-12-21:
@actuallymentor Just following up, when I typed `battery status` in Terminal, this is the 1st line:

`touch: /Users/praveen/.battery/battery.log: Permission denied`

Giving Battery full disk access in Privacy & Security does not fix this.

How can I give Battery permission to write to the battery.log file?

IB3N, 2023-01-16:
> @actuallymentor Just following up, when I typed `battery status` in Terminal, this is the 1st line:
>
> `touch: /Users/praveen/.battery/battery.log: Permission denied`
>
> Giving Battery full disk access [...]

I'm also having this Permission denied issue when writing to battery log