actuallymentor/battery #28: Using the CLI version, it doesn't seem to work in clamshell mode.

svk-works, 2022-11-22:


svk-works, 2022-11-24:
[image]
It doesn't seem to work even in normal mode :(

actuallymentor, 2022-12-12:
Can you show me a trail of `tail ~/.battery/battery.log`?

DUOLabs333, 2022-12-20:
This is because the system sleeps, so `battery` can't toggle charging as needed. I believe AlDente gets around this having a sleep inhibit assertion.

actuallymentor, 2023-01-05:
@DUOLabs333 that seems likely. I'm not sure how happy I am with the idea of forcing wakefulness... Do you know if OSX has some sort of `pre sleep` event listener I could hook into that is triggered before the device goes to sleep?

DUOLabs333, 2023-01-05:
You could just disable sleep entirely with `sudo pmset -a disablesleep 1`.

actuallymentor, 2023-01-05:
> You could just disable sleep entirely with `sudo pmset -a disablesleep 1`.

I'm not sure I'm comfortable with doing that as it can cause battery drain.

DUOLabs333, 2023-01-05:
There are sleep listeners, but they require you to use the C API.

DUOLabs333, 2023-01-05:
Alternatively, you can detect when it is in Clamshell mode (which can be done on the command line), then count out whatever time you set before it goes to sleep, then do whatever you need to do before then.

volodya7292, 2023-02-08:
On the latest version of the CLI another cause is discharging to the target level before maintaining the battery level. Triggering discharge disables the adapter, but clamshell mode works only when the adapter is enabled.

waterlou, 2023-02-08:
Just updated to 1.0.6 and now clamshell mode suddenly not working, uninstall battery app solved the problem.  M1Max MacBook Pro 14, Ventura 13.2

UsamaKarim, 2023-02-12:
> Just updated to 1.0.6 and now clamshell mode suddenly not working, uninstall battery app solved the problem. M1Max MacBook Pro 14, Ventura 13.2
> 
> FYI, downgraded to 1.0.5 fixed the problem.

Not working for me on M1 MBA

volodya7292, 2023-02-12:
@UsamaKarim I got it working by modifying `setup.sh` and downgrading the CLI to revision 2f3a243. Modified setup.sh: https://gist.github.com/volodya7292/218fced5070c441720ae76434b55904f

Nick-Z0, 2023-02-15:
Can confirm this works. Macbook 14 (2021) MacOS Ventura. 
I had it working fine before on Monterey, but after the update to Ventura it broke i think. I had installed it through brew on Monterey, but as explained in another issue, the brew version is old (I think it was v1.03).

So simple instructions for temporary workaround:

0. (Don't know if this is needed, but I did it) Quit app and uninstall previous version?: 
`sudo rm -rf ~/.battery
binfolder=/usr/local/bin
sudo rm -v "$binfolder/smc" "$binfolder/battery"`
1. Go to repo home, download as .zip
2. Edit setup.sh with text editor, replace all content with the contents of this and save file: https://gist.github.com/volodya7292/218fced5070c441720ae76434b55904f
3. Open terminal, navigate to the folder you extracted the .zip file.
4. Run `bash setup.sh`, enter password.
5. ????
6. Profit!!!

2001Y, 2023-02-24:
I am the same way. In clamshell mode, the external screen is turned off.

sataset, 2023-02-25:
For me `battery charge X` gets the job done in clamshell mode.

`battery maintain X` disables adapter thus putting laptop to sleep.

MacBook Pro 14, macOS Monterey 12.6.3 (21G419)

P.S. Did not experience this issue previously, do now after update.

j4r3kb, 2023-03-07:
You can use an app like Amphetamine to prevent sleep in various scenarios.

actuallymentor, 2023-03-08:
Reading up on this again and I'm thinking solving this would require two things:

1. Detect clamshell mode (possibly on every command run)
2. Not doing discharge forcing in clamshell (since it shuts down the external screen as I understand from this thread)
3. Manually disabling sleep while in clamshell mode (possibly only when adapter is not plugged in?)
4. Putting a warning label in the Readme to explain the caveats

I'm not sure when I'm going to make some more time to work on `battery` so will mark this as `pr welcome`

schabrolles, 2023-03-08:
As I said in #59 I also found a quick workaround.
---

I think this is because of the "discharging mode" which is enabled by default when `battery maintain 80` is used...

I just disable it in the script `/usr/local/bin/battery` (by commenting line 348)

```
346         # Before we start maintaining the battery level, first discharge to the target level
347         log "Triggering discharge to $setting before enabling charging limiter"
348         #battery discharge "$setting"
349         log "Discharge pre battery-maintenance complete, continuing to battery maintenance loop"
```

Now everything seems to be OK for me ... I can use my external monitor with thunderbolt charging.
- MacBook is powered even when battery is >80% but **battery is not charging** (this avoid laptop to sleep when lid is closed) 
- Battery is not discharging when connected to power and battery level >80% (which is what we want)
- Battery is charging when when connected to power and battery level <80% (which is what we want)

@actuallymentor, Do you think you can add a function to avoid battery discharging when plugged ? This should easily resolve that external monitor issue.

gekoxyz, 2023-03-11:
> ## As I said in #59 I also found a quick workaround.
>
> I think this is because of the "discharging mode" which is enabled by default when `battery maintain 80` is used...
>
> I just disable it in the [...]

This fixed it for me. The most important thing if i'm with more than 80% battery in clamshell is to not stress the battery (keep it charged) so i think unless there's a way to go to clamshell mode with unplugged battery (like with Amphetamine app) this is the way to go

actuallymentor, 2023-03-12:
@schabrolles I've toyed around a bit and changed the `caffeinate` settings to be more agressive. Could you try with the latest `battery` version (including the CLI) and UNCOMMENT the bit you did to see if this fixed it for you?

schabrolles, 2023-03-12:
On which version ? 

I'm running v1.0.3 for the moment

actuallymentor, 2023-03-12:
Ah that might be why I'm unable to reproduce it, my monitor always gives power too.

I'll look into it 👍

schabrolles, 2023-03-13:
Just to be clear... when connected to an external monitor in clamshell mode, we need to always let power going to the Mac (to avoid the forced sleep) but disable the charging mode when <80% (to let the battery rest and stay at the same level)

sataset, 2023-03-13:
I just realized why it was working before.

Clamshell + maintain X works fine if battery level is below X, because it stops charging at that level, but still provides adapter power.

However if battery have more than X charge, then maintain 80 disables the adapter and attempts to discharge the battery which puts laptop to sleep.

@actuallymentor possible solution is to have extra option for `maintain` command. E.g. `battery maintain 80 --force-disharge`. As soon battery is not charged it is totally fine to keep it as is and sip power from the adapter.

actuallymentor, 2023-03-13:
@sataset and @schabrolles I disabled the force discharge by default and will be adding a GUI option for it. Please let me know whether it works as intended for you now.

actuallymentor, 2023-03-13:
Clamshell users, please try [this beta release](https://github.com/actuallymentor/battery/releases/tag/v1.1.4-beta) and thumbs up/down it to let me know if it works for you now!

There is now a new GUI option:

[image]

pkonopelko, 2023-03-13:
I confirm the new feature works great! No need to comment out the line @schabrolles mentioned anymore. Many thanks! I also tested that after enabling "allow force discharging" works as intended, i.e. exactly like previously.

By the way @actuallymentor thank you so much for your time and efforts put into developing this excellent tool which **battery** is! Thank you for your great work.

schabrolles, 2023-03-13:
This works like a charm ... 👍🏻

"For Fun" I try to force "Allow discharging mode" ... I had the nice surprise to see a pop up message telling me that it can put my laptop to sleep in clamshell mode .... And Yes ... It does 😅 ...

Thanks for your work

schabrolles, 2023-03-13:
@actuallymentor I also learn today how I can avoid my Mac to goes to sleep while closing the lid in battery mode (this help me to keep my job running and not cutting my network connection when walking to the meeting room...

"pmset disablesleep 1"  and  "pmset disablesleep 0"  may be an Idea to add into advanced settings .... or not (you decide)

sataset, 2023-03-14:
Confirming this works for me too.

Joining to pkonopelko's words and want to thank you for creating and investing your time to project ❤️

schabrolles, 2023-03-14:
tonight, my screen automatically "wake up" every minutes ... (so me too 😒) ... 
@actuallymentor, could it be due to `caffeinate`? (just a guess)... 
Honestly, I was to tired to test last night (easiest solution: I powered off my Mac 😉). 
Does anybody notice the same issue ?

actuallymentor, 2023-03-14:
Closed by release [v1.1.4](https://github.com/actuallymentor/battery/releases/tag/v1.1.4). Please reopen if you have the same issue with that release (with debug info).

@schabrolles I suspect you have an older version, there was an intermediate version where I tried to fix the clamshell issue by forcing the device to stay awake. I suspect that is what was happening to you.

closed 2023-03-14 by actuallymentor