actuallymentor/battery #30: MacOS 13 M1 Not in effect

almightyYantao, 2022-11-30:
[image]

almightyYantao, 2022-11-30:
[image]

love4taylor, 2022-12-02:
```
~
❯ battery status
18:03:08 - Battery at 80% (attached; remaining), smc charging disabled
18:03:08 - Your battery is currently being maintained at 80%
```
[image]

[image]

almightyYantao, 2022-12-06:
[image]
[image]
[image]

safferli, 2022-12-12:
you will need to disable the MacOS-specific automatic battery monitoring, I believe

actuallymentor, 2022-12-12:
I do not speak Chinese I'm afraid, did you solve the issue?

closed 2022-12-14 by actuallymentor