actuallymentor/battery #34: Error handling when running without sudo

Tokarak, 2022-12-12:
`sudo` is required to run most battery commands. Currently the output without sudo is:
```
12:14:12 - Starting battery maintenance at 75%
/usr/local/bin/battery: line 305: /Users/tok/.battery/battery.pid: Permission denied
/usr/local/bin/battery: line 307: /Users/tok/.battery/maintain.percentage: Permission denied
12:14:12 - Maintaining battery at 75%
```
It is ambiguous wether the command succeeded. I'm assuming superuser permissions are required, otherwise less permissions would be less secure. Code to check for the UID 0 bit should be added, and a help message should be printed.
I may be happy to open a PR when I have time.

actuallymentor, 2022-12-14:
You are not supposed to run the command with sudo :)

If you do this, the log and pid files might get root ownership and break things. Could you try force-uninstalling with:

```shell
battery uninstall
sudo rm -rf ~/.battery
binfolder=/usr/local/bin
sudo rm -v "$binfolder/smc" "$binfolder/battery"
```

And then reinstalling with your preferred method?

Tokarak, 2023-01-23:
Yes, thank you. pid and log file got root ownership, like you said.

closed 2023-01-23 by Tokarak