actuallymentor/battery #36: It just doesn't work on my M1

data-science-made-easy, 2022-12-13:
I figured out your script uses 'smc' under the hood. The two commands `smc -k CH0B -w 02` and `smc -k CH0C -w 02` should do the trick and stop the charging, if I understand battery.sh correctly. Running these two commands however don't seem to do anything :(

Can you explain why you think the exact above two commands should stop the charging? Any idea what can be wrong?

Thanks!

actuallymentor, 2022-12-14:
Are you trying to use this tool, or the underlying smc tool for other reasons?

What exactly is not working? What errors do you see?

kramnda1, 2023-01-08:
Same issue here on my M1 Pro 16GB running Ventura 13.1.

[image]

For good luck I tried installing with brew, with the dmg package and lastly form source. CLI commands just do not seem to do anything. Also tried using the underlying `smc` commands, still no luck. I do not get any errors.

`tail ~/.battery/battery.log` looks like this:

>01/08/23-10:09:07 - Charging to and maintaining at 80% from 100%
>01/08/23-10:27:11 - Charging to and maintaining at 80% from 100% 
>01/08/23-10:27:11 - Charge above 80 
>01/08/23-10:27:11 - Disabling battery charging

Log is after fresh install with brew, running the app and using the commands below (tried the same thing after restart). 

`battery charging off`
`battery status`
`battery charging on`
`battery status`
`battery maintain 80`
`battery charging off` 

The battery status (from the screenshot) just remains the same whether I turn charging on/off with the CLI commands.
[image]

I am using external monitor that I use as dock with usc c charging. I tried using just a magsafe charger (without the monitor plugged in) just to try it out, but still nothing. 

I do not think it is just the status bar issue, seems to me like `smc` related issue, but I do not know anything about it.

Thank you!

csjondev, 2023-01-15:
I am having the same issue as well on the same m1 mac

anarchy89, 2023-01-15:
Did you guys try to let the battery drop a little to say like 95%, then run the command and plug it back in?

csjondev, 2023-01-15:
> Did you guys try to let the battery drop a little to say like 95%, then run the command and plug it back in? It works fine for me.

Yeah, nothing. I don't know if it makes a big difference but I mainly use my M1 macbook air in clamshell mode since I have external monitors

anarchy89, 2023-01-15:
> > Did you guys try to let the battery drop a little to say like 95%, then run the command and plug it back in? It works fine for me.
> 
> 
> 
> Yeah, nothing. I don't know if it makes a big difference but I mainly use my M1 macbook air in clamshell mode since I have external monitors

Well did you try it without clamshell mode first?

kramnda1, 2023-01-15:
Well it appears to work for me now after letting the battery drop a bit (had 82% when I tested it). It just took couple second to register after I used the command `battery charging off`. It also works in clamshell mode. I do not know why it did not work beforehand, but in the meantime I installed Aldente. If it still does not work for you, you can try to install Aldente, uninstall it and see if it works 😄, but I admit this is kinda dumb thing to try...

anarchy89, 2023-01-15:
> Well it appears to work for me now after letting the battery drop a bit (had 82% when I tested it). It just took couple second to register after I used the command `battery charging off`. It also [...]

Yeah it takes a few seconds, you can also unplug the wire and plug it back I that works too if you can't wait. Not a fan of Al dente.

csjondev, 2023-01-15:
Weird I had to let the battery run down to 84% and now it works. Maybe this is a bug fix he can work on

anarchy89, 2023-01-15:
> Weird I had to let the battery run down to 84% and now it works. Maybe this is a bug fix he can work on

I don't think it's a bug. It just displays differently when the battery if fully charged.

st-ivan, 2023-01-19:
> Weird I had to let the battery run down to 84% and now it works. Maybe this is a bug fix he can work on

ok, im having the same issue, clamshell mode all the time. Im at 100% right now and it doesnt seem to do anything. 
Im going to run it down to 70 and then plug it back in. Then it should charge it to 80% and keep it there forever right? Is that how its working for you?

anarchy89, 2023-01-19:
> ok, im having the same issue, clamshell mode all the time. Im at 100% right now and it doesnt seem to do anything.
>
> Im going to run it down to 70 and then plug it back in. Then it should charge it [...]

It doesn't matter. Just run it down to any number that's not 100. After that it will slowly drain. Once a month you should drain the battery to 20-40% and charge it back to 80ish. Then do the same thing.

st-ivan, 2023-01-20:
hey there, thanks for the info, i did a calibration and plugged it back at 75%, enabled limit to 80% and it charged it up to it and now it says "On Power Adapter". I guess its working as intended.

actuallymentor, 2023-02-02:
From what I'm reading the macOS interface doesn't update if the battery is entirely full. This sounds related to the #20 issue.

I'll close this for now with the note that a future version might include a discharge functionality.

closed 2023-02-02 by actuallymentor

anarchy89, 2023-02-02:
> From what I'm reading the macOS interface doesn't update if the battery is entirely full. This sounds related to the #20 issue.
> 
> 
> 
> I'll close this for now with the note that a future version might include a discharge functionality.

Hi, if you do that please don't change the logic of battery charging off, it does exactly what it's supposed to do.