actuallymentor/battery #37: Installation fails with error "command not found"

rubenet, 2022-12-14:
Hi

first of all thanks for creating and sharing this.

I suspect my error might be related to the other open issue #16", but not sure. 

I am using an Apple M1 Pro with Ventura 13.0.1.

I got these errors on installation via the DMG file, and some errors later when I tried the installation via command line:

```
- Error installing battery limiter: undefined
- Battery limiter error: Command failed: PATH=$PATH:/usr/local/bin/battery battery status
/bin/bash: battery: command not found
```

Then when I tried to click "Enable 80% battery limit:

```
- Command failed: PATH=$PATH:/usr/local/bin/battery battery maintain 80
/bin/bash: battery: command not found
```

Without uninstalling, I tried to install via the command line:

```
secret@machine ~ % curl -s https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | bash
####################################################################
# 👋 Welcome, this is the setup script for the battery CLI tool.
# Note: this script will ask for your password once or multiple times.
####################################################################

Password:
🔋 Starting battery installation

[ 1/9 ] Superuser permissions acquired.
[ 2/9 ] Cloning fan control version of smc
bash: line 30: cd: /Users/secret/.battery-tmp/smc/smc-command: No such file or directory
[ 3/9 ] Building smc from source
[ 4/9 ] Move smc to executable folder
mv: rename /Users/secret/.battery-tmp/smc/smc-command/smc to /usr/local/bin/smc: No such file or directory
chmod: /usr/local/bin/smc: No such file or directory
[ 5/9 ] Cloning battery repository
[ 6/9 ] Writing script to /usr/local/bin/battery for user secret
cp: /Users/secret/.battery-tmp/battery/battery.sh: No such file or directory
chown: /usr/local/bin/battery: No such file or directory
chmod: /usr/local/bin/battery: No such file or directory
chmod: /usr/local/bin/battery: No such file or directory
chown: /usr/local/bin/battery: No such file or directory
bash: /Users/secret/.battery-tmp/battery/battery.sh: No such file or directory
[ 7/9 ] Set up visudo declarations
[ 8/9 ] Removing temp folder /Users/secret/.battery-tmp
[ 9/9 ] Removed temporary build files

🎉 Battery tool installed. Type "battery help" for instructions.

secret@machine ~ % battery maintain 80
zsh: command not found: battery

```
I manually created these two folder with "sudo mkdir", but still got errors in the installation:

/usr/local/bin/smc
/usr/local/bin/battery:

```
secret@machine ~ % curl -s https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | bash

####################################################################
# 👋 Welcome, this is the setup script for the battery CLI tool.
# Note: this script will ask for your password once or multiple times.
####################################################################

🔋 Starting battery installation

[ 1/9 ] Superuser permissions acquired.
[ 2/9 ] Cloning fan control version of smc
bash: line 30: cd: /Users/secret/.battery-tmp/smc/smc-command: No such file or directory
[ 3/9 ] Building smc from source
[ 4/9 ] Move smc to executable folder
mv: /Users/secret/.battery-tmp/smc/smc-command/smc: No such file or directory
[ 5/9 ] Cloning battery repository
[ 6/9 ] Writing script to /usr/local/bin/battery for user secret
cp: /Users/secret/.battery-tmp/battery/battery.sh: No such file or directory
bash: /Users/secret/.battery-tmp/battery/battery.sh: No such file or directory
[ 7/9 ] Set up visudo declarations
[ 8/9 ] Removing temp folder /Users/secret/.battery-tmp
[ 9/9 ] Removed temporary build files

🎉 Battery tool installed. Type "battery help" for instructions.
```

Any help is appreciated. 

Thanks,
Ruben

actuallymentor, 2022-12-17:
Could you run this and show me the output?

```bash
tempfolder=~/.battery-tmp
smcfolder="$tempfolder/smc"
rm -rf $smcfolder
git clone --depth 1 https://github.com/hholtmann/smcFanControl.git $smcfolder
cd $smcfolder/smc-command
make
```

rubenet, 2022-12-18:
Hi,

I ran your lines and got an error, found I was missing "xcode" so I installed it and rerun your lines again, which seemed to succeed although with a few warnings. See below. 
After that it seems the app works for me, I can toggle the enable/disable without errors and got the message "Battery background component installed successfully" when I first ran the app.

Now I will monitor the charge but I think this is how it looks when the "charge limit" is in place and charging is on hold:

[image]

And I have just seen the battery level going down to 98% so looking good. 

Thanks for the help and for sharing this app!

```
 	
secret@machine ~ % 
tempfolder=~/.battery-tmp &&
smcfolder="$tempfolder/smc" &&
rm -rf $smcfolder &&
git clone --depth 1 https://github.com/hholtmann/smcFanControl.git $smcfolder &&
cd $smcfolder/smc-command &&
make

xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun
secret@machine ~ % xcode-select --install
xcode-select: note: install requested for command line developer tools
secret@machine ~ %                       
tempfolder=~/.battery-tmp &&
smcfolder="$tempfolder/smc" &&
rm -rf $smcfolder &&
git clone --depth 1 https://github.com/hholtmann/smcFanControl.git $smcfolder &&
cd $smcfolder/smc-command &&
make

Cloning into '/Users/secret/.battery-tmp/smc'...
remote: Enumerating objects: 331, done.
remote: Counting objects: 100% (331/331), done.
remote: Compressing objects: 100% (174/174), done.
remote: Total 331 (delta 121), reused 251 (delta 102), pack-reused 0
Receiving objects: 100% (331/331), 1.42 MiB | 2.14 MiB/s, done.
Resolving deltas: 100% (121/121), done.
gcc -DCMD_TOOL_BUILD -c smc.c
smc.c:37:1: warning: 'OSSpinLock' is deprecated: first deprecated in macOS 10.12 - Use os_unfair_lock() from <os/lock.h> instead [-Wdeprecated-declarations]
OSSpinLock g_keyInfoSpinLock = 0;
^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/libkern/OSSpinLockDeprecated.h:79:17: note: 'OSSpinLock' has been explicitly marked deprecated here
typedef int32_t OSSpinLock OSSPINLOCK_DEPRECATED_REPLACE_WITH(os_unfair_lock);
                ^
smc.c:294:2: warning: 'IOMasterPort' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
        IOMasterPort(MACH_PORT_NULL, &masterPort);
        ^~~~~~~~~~~~
        kIOMainPortDefault
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:143:1: note: 'IOMasterPort' has been explicitly marked deprecated here
IOMasterPort( mach_port_t       bootstrapPort,
^
smc.c:346:5: warning: 'OSSpinLockLock' is deprecated: first deprecated in macOS 10.12 - Use os_unfair_lock_lock() from <os/lock.h> instead [-Wdeprecated-declarations]
    OSSpinLockLock(&g_keyInfoSpinLock);
    ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/libkern/OSSpinLockDeprecated.h:99:9: note: 'OSSpinLockLock' has been explicitly marked deprecated here
void    OSSpinLockLock( volatile OSSpinLock *__lock );
        ^
smc.c:379:5: warning: 'OSSpinLockUnlock' is deprecated: first deprecated in macOS 10.12 - Use os_unfair_lock_unlock() from <os/lock.h> instead [-Wdeprecated-declarations]
    OSSpinLockUnlock(&g_keyInfoSpinLock);
    ^
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/libkern/OSSpinLockDeprecated.h:105:9: note: 'OSSpinLockUnlock' has been explicitly marked deprecated here
void    OSSpinLockUnlock( volatile OSSpinLock *__lock );
        ^
smc.c:395:23: warning: format string is not a string literal (potentially insecure) [-Wformat-security]
    sprintf(val->key, key);
                      ^~~
[... 16 lines cut ...]
smc.c:668:22: note: treat the string as an argument to avoid this
    sprintf(val.key, key);
                     ^
                     "%s", 
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/secure/_stdio.h:47:56: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                                                       ^
smc.c:778:34: warning: format string is not a string literal (potentially insecure) [-Wformat-security]
                sprintf(val.key, key);
                                 ^~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/secure/_stdio.h:47:56: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                                                       ^~~~~~~~~~~
smc.c:778:34: note: treat the string as an argument to avoid this
                sprintf(val.key, key);
                                 ^
                                 "%s", 
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/secure/_stdio.h:47:56: note: expanded from macro 'sprintf'
  __builtin___sprintf_chk (str, 0, __darwin_obsz(str), __VA_ARGS__)
                                                       ^
7 warnings generated.
gcc -mmacosx-version-min=10.4  -Wall -g -framework IOKit -o smc smc.o
ld: warning: object file (smc.o) was built for newer macOS version (13.0) than being linked (11.0)

```

rubenet, 2022-12-19:
secret@machie ~ % battery status
12:51:14 - Battery at 87% (attached; remaining), smc charging disabled
12:51:14 - Your battery is currently being maintained at 80%
secret@machine ~ % 

Working for me now, thank you!

closed 2022-12-19 by rubenet