actuallymentor/battery #39: Issue with GUI and CLI "sudo: smc: command not found"

aroonjawaid, 2022-12-28:
Hi,

I'm having some issues with both battery GUI and CLI.

On the GUI after installing and downloading the  components, I get the following prompt,  "Error installing battery limiter: undefined"

With the CLI, whenever I run a command, I get this error.

sudo: smc: command not found sudo: smc: command not found
/us/local/bin/battery: line 110: smc: command not found

jju1ez, 2022-12-30:
For me it is the following error message on the GUI:
[image]

actuallymentor, 2023-01-05:
@aroonjawaid can you send me the output of: `tail ~/.battery/battery.log` please?

actuallymentor, 2023-01-05:
@jju1ez same for you :)

ffasbend, 2023-01-25:
You must have the Xcode command line tools installed, when installing the program. The program fetches code from Github and needs the git command, which is not installed per default on MAC.
Open Terminal and install Xcode command line tools first with: xcode-select --install 
(Perhaps a remark can be added on the Readme file)

actuallymentor, 2023-02-02:
@ffasbend holy sh*t I had never considered that people might not have git installed! I'll have a look at adding a "install git if not installed" section to the setup script.

actuallymentor, 2023-02-04:
I just pushed a update that should fix a few issues, could you try to run `brew reinstall --cask battery` and see if that fixes it?

actuallymentor, 2023-02-05:
Closing under the assumption the most recent release fixes this. Please re-open if this is not the case.

closed 2023-02-05 by actuallymentor

earthsound, 2023-02-14:
> I just pushed a update that should fix a few issues, could you try to run `brew reinstall --cask battery` and see if that fixes it?

I'm running v1.0.8 (installed via homebrew on an M1 Pro running macOS 12.6.2 and updated today) and, although I'm not having the undefined error the OP stated nor do I see an error when running battery commands in the terminal, I do see the sudo: smc errors in the Battery CLI Logs. I have Xcode command line tools installed (xcode-select version 2395).

E.g.:

02/14/23-01:28:21 - Charge below 80
02/14/23-01:28:21 - 🔌🔋 Enabling battery charging
02/14/23-01:28:21 - 🔼🪫 Disabling battery discharging
02/14/23-10:02:50 - Charge above 80
02/14/23-10:02:50 - 🔌🪫 Disabling battery charging
02/14/23-11:35:07 - Charge below 80
02/14/23-11:35:07 - 🔌🔋 Enabling battery charging
02/14/23-11:35:07 - 🔼🪫 Disabling battery discharging
02/14/23-13:27:59 - Triggering discharge to 80 before enabling charging limiter
02/14/23-13:27:59 - Discharging to 80% from 60%
02/14/23-13:27:59 - 🔽🪫 Enabling battery discharging
sudo: smc: command not found
02/14/23-13:27:59 - 🔼🪫 Disabling battery discharging
sudo: smc: command not found
02/14/23-13:27:59 - Discharging completed at 60%
02/14/23-13:27:59 - Discharge pre battery-maintenance complete, continuing to battery maintenance loop
02/14/23-13:27:59 - Charging to and maintaining at 80% from 60%
/usr/local/bin/battery: line 143: smc: command not found
02/14/23-13:27:59 - Charge below 80
02/14/23-13:27:59 - 🔌🔋 Enabling battery charging
sudo: smc: command not found
sudo: smc: command not found
02/14/23-13:27:59 - 🔼🪫 Disabling battery discharging
sudo: smc: command not found
02/14/23-13:28:34 - Triggering discharge to 80 before enabling charging limiter
02/14/23-13:28:34 - Discharging to 80% from 60%
02/14/23-13:28:34 - 🔽🪫 Enabling battery discharging
02/14/23-13:28:34 - 🔼🪫 Disabling battery discharging
02/14/23-13:28:34 - Discharging completed at 60%
02/14/23-13:28:34 - Discharge pre battery-maintenance complete, continuing to battery maintenance loop
02/14/23-13:28:34 - Charging to and maintaining at 80% from 60%

Output running smc in terminal:

```
$ smc -h
Apple System Management Control (SMC) tool 0.01
Usage:
smc [options]
    -f         : fan info decoded
    -t         : list all temperatures
    -h         : help
    -k <key>   : key to manipulate
    -l         : list all keys and values
    -r         : read the value of a key
    -w <value> : write the specified value to a key
    -v         : version
```