actuallymentor/battery #40: How is the log file written to after reboot?

whiterock, 2022-12-31:
This question might seem a bit silly, but hear me out. So I read the code and when `battery maintain` is called in turn 
```
nohup battery maintain_synchronous $setting >> $logfile &
```
Now this spawns a process with writes to the log file. So far so good. Now what happens when I reboot? Then the LaunchAgent gets executed which calls
```
/usr/local/bin/battery maintain_synchronous recover
```
spawning a process, which is NOT piped to the logfile. But now the strangest things happen. There is stuff written to the logfile on reboot anyway. I rebooted at 0:41 and got
```
18:17:57 - Charging to and maintaining at 79% from 91%
00:41:35 - Charge below 79
00:41:35 - Enabling battery charging
sudo: you do not exist in the passwd database
sudo: you do not exist in the passwd database
```
I am totally confused for three reasons:
1. How can it write to the logfile when battery.sh is just full of echos and the .plist does not pipe to the logfile
2. My battery percentage was 85% at reboot, so the log entries are false in that sense. But they are also false in another sense, since when I run `battery status` it behaves correctly:
```
❯ battery status
00:45:13 - Battery at 87% (11:33 remaining), smc charging disabled
00:45:13 - Your battery is currently being maintained at 79%
```
3. Where does the `sudo: you do not exist in the passwd database` come from?

So all in all it does seem to behave correctly in maintaining the charge limit, but the logging is odd (and I cannot fathom how they even happen), the logs are false and there is a sudo error in them.

thanks in advance, I will provide you all further info you need :)

*Further info*: I did the command-line only installation and have an M1 Pro machine from 2021.

actuallymentor, 2023-01-05:
I love this question @whiterock, and I'm confused as well 😅. Follow-ups:

Do you have the GUI installed as well? Does it launch at boot? This would explain things since the GUI runs `battery maintain` on start which would override the daemon.

I've updated the binary to do some extra logging (see [this commit](https://github.com/actuallymentor/battery/commit/2f3a243cc2d23e8c6dab5729c4f84bd6b81978d9)). Could you let me know what that shows?

whiterock, 2023-01-06:
Okay so I did the update and then did two restarts. Here is what I got:
```
01/06/23-11:29:52 - Charging to and maintaining at 59% from 70%
01/06/23-11:30:52 - Charge below 59
01/06/23-11:30:52 - Enabling battery charging
sudo: you do not exist in the passwd database
sudo: you do not exist in the passwd database
01/06/23-11:31:45 - Debug trail. User: whiterock, config folder: /Users/whiterock/.battery, logfile: /Users/whiterock/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
01/06/23-11:31:45 - Recovering maintenance percentage 59
01/06/23-11:31:47 - Charging to and maintaining at 59% from 69%
01/06/23-11:31:47 - Charge above 59
01/06/23-11:31:47 - Disabling battery charging
01/06/23-13:58:29 - Debug trail. User: whiterock, config folder: /Users/whiterock/.battery, logfile: /Users/whiterock/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
01/06/23-13:58:29 - Recovering maintenance percentage 59
01/06/23-13:58:29 - Charging to and maintaining at 59% from 71%
01/06/23-13:58:29 - Charge above 59
01/06/23-13:58:29 - Disabling battery charging
```
Things to note:
- For the restart at 11:31:?? (I did not write it down, it shows this time in `last reboot` in zsh) the program was running as `nohup ...` after calling `battery maintain 59`
- For the restart at 13:57:50 (I wrote the shutdown time down here exactly) the program was running by means of launchctl as intended from the previous restart

Now as it seems the second restart worked perfectly. My suspicion is, that the log lines containing wrong information and the the `sudo` lines occur at shutdown maybe? For the sudo lines it's hard to tell, but `last reboot` suggests I rebooted at 11:31, so that would be before that, but I don't know the precision or the exact time this stuff is measured at. However logging in s.t. the LaunchAgent runs does take almost a minute, so yes, I suspect it is being written at shutdown - maybe `pmset -g batt` shows 0 or something like that when being shut down. The sudo lines, no idea.

ad GUI:
I did install the GUI briefly, only because `brew install battery` errored with `formulae not found`, but quickly uninstalled it since I am not much a GUI person, so this should not be an issue. I then installed it the manual method. I'd actually prefer to install it via `brew`, since I keep everything organized that way, maybe you get around to fixing that some time :)

I would consider the bug more or less fixed - Logging certainly works now with the updated .plist. 👍🏼

actuallymentor, 2023-01-07:
> I did install the GUI briefly, only because brew install battery errored with formulae not found,

Did you run a `brew update` before installing? The brew version installs the GUI which installs the CLI under the hood.

> I would consider the bug more or less fixed - Logging certainly works now with the updated .plist. 👍🏼

Good to hear, I'll close this one for now then :)

closed 2023-01-07 by actuallymentor