actuallymentor/battery #43: Not Working

csjondev, 2023-01-15:
Battery continues to charge at 100%. Even in iTerm2 when i run battery charging off it says it is disabling battery charging but upon checking it does not say it is not charging anymore

st-ivan, 2023-01-19:
new user here. installed ,battery at 100% it shows lighting icon. wonder if this app works at all in clamshell mode. I tried aldente and it doesnt work either (they claim it does via pro mode but it doesnt)

Zeroscl, 2023-01-26:
I followed all the indications for the installation of the app, then I used the laptop (MB Pro M1) until the battery ran out at 25% and when I connected it it continued charging up to 100%, it did not stop charging at 80%, any recommendations? Thanks

st-ivan, 2023-01-26:
> I followed all the indications for the installation of the app, then I used the laptop (MB Pro M1) until the battery ran out at 25% and when I connected it it continued charging up to 100%, it did not stop charging at 80%, any recommendations? Thanks

hey there, I plugged it back at 75%, enabled limit to 80% and it charged it up to it and now it says "On Power Adapter".

actuallymentor, 2023-02-04:
I just pushed a update that should fix a few issues, could you try to run `brew reinstall --cask battery` and see if that fixes it?

actuallymentor, 2023-02-05:
Closing under the assumption that the most recent release fixes this. Please re-open if this is not the case.

closed 2023-02-05 by actuallymentor