actuallymentor/battery #44: The battery level setting is preserved, but it still charges until I run battery again

ballo, 2023-01-22:
**What is the issue? (required)**
If I reboot (or some other reset situation?) the battery status shows the level of the battery to be maintained has been preserved, but the charging is enabled even if the battery level is much higher until I run battery maintain. 

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. sudo battery maintain 40
2. reboot
3. sudo battery status
5. See "Your battery is currently being maintained at 40%" which is correct (battery level preserved) but also nonsensically see "smc charging enabled"
6. sudo battery maintain 40
7. sudo battery status
8. See "Your battery is currently being maintained at 40%" but also "smc charging disabled" which is obviously what I want

**Expected behavior (required)**
If the battery level is above what the maintenance level is, charging should be disabled. 

**Screenshots (optional)**
[image]

ballo, 2023-01-22:
BTW this is on a 8 core m1 13" macbook pro

ballo, 2023-02-01:
Sigh, it reset this time and I didn't even reboot. Back on 100%.

I constantly have to babysit this thing. I might have to create a cron job to periodically check the battery level to detect when this is malfunctioning

actuallymentor, 2023-02-05:
Can you show me the output of `battery logs` before you reboot and after? It would really help in debugging.

ballo, 2023-02-09:
I've never seen battery logs show any output

ballo, 2023-03-14:
[image]
It happened again. I think this time because I had to reboot

closed 2023-07-20 by actuallymentor