actuallymentor/battery #47: [Feature Request] Improve Menu GUI information

NdR91, 2023-01-30:
**Is your feature request related to a problem? Please describe.**
The GUI menu only shows a standard set of options:

- Enable 80% Limit;
- Disable 80% Limit;

[image]

**Describe the solution you'd like**
Let the Top bar menu show the setup done in the CLI.
For example the informations I get with the `battery status`command

A nice set of information to be displayed in that menu is:

- Mantain percentage set;
- Charging Enabled/Disabled;
- Battery charging YES or NOT;

**Describe alternatives you've considered**
The alternative is to get those info inside the terminal. Not properly comfortable.

ballo, 2023-02-01:
I don't even use the menu. Maybe fix the basic functionality of the program first so it doesn't keep charging past the selected level.

NdR91, 2023-02-01:
> I don't even use the menu. Maybe fix the basic functionality of the program first so it doesn't keep charging past the selected level.

This comment actually don't contribute on this issue. 
Open a new one (or +1 an existing one) if you need some bug fix.

JacksonChen666, 2023-02-05:
@actuallymentor Keywords to close an issue does not function in GitHub releases. It is meant to be used in commit messages.

actuallymentor, 2023-02-05:
@JacksonChen666 I see you're very on top of things, thanks ;)

closed 2023-02-05 by actuallymentor

NdR91, 2023-02-05:
> @JacksonChen666 I see you're very on top of things, thanks ;)

Really thank you for your work!