actuallymentor/battery #48: Not working for M2 Max MacBook Pro 16 inch

tylermilner, 2023-02-02:
**What is the issue? (required)**
The app fails to launch on my M2 Max MacBook Pro with the following error:

```
Error installing battery limiter: Command failed: ./applet/bin/sh: ./applet: Bad CPU type in executable
```

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Install `battery` via Homebrew: `brew install battery`. Installation succeeds.
2. Open `/Applications/Battery.app`.
3. Observe "Bad CPU type in executable" error.

**Expected behavior (required)**
The application should launch without error.

**Screenshots (optional)**
[image]

**Error logs (optional)**
The log for `~/.battery/battery.log` does not exist. Here is the log for `~/.battery/gui.log`:

```
Starting tray app
Tray app boot complete
Triggering boot-time auto-update
Is installed? 
false
Installing battery for tylermilner...
Update/install error: 
Error: Command failed: ./applet
/bin/sh: ./applet: Bad CPU type in executable

Update process complete
Error getting battery status: 
Error: Command failed: PATH=$PATH:/usr/local/bin/battery battery status
/bin/bash: battery: command not found
```

**Additional context (optional)**
Add any other context about the problem here.

actuallymentor, 2023-02-02:
Could you try running `xcode-select --install` and then opening the app again?

tylermilner, 2023-02-03:
Running `xcode-select --install` resulted in the following error:

```
xcode-select: error: command line tools are already installed, use "Software Update" in System Settings to install updates
```

However, opening up Xcode and just reselecting the option seems to have worked:
[image]

The app is opening now. Thanks for your help!

tylermilner, 2023-02-03:
So it does look like the app opens now, but I still get the error messages.

First I get this screen, but nothing prompts me to input my password or anything after clicking "OK":
[image]

Then I get the same two errors as before:
[image]

[image]

After that, the app does look to be running in the background, but I'm wondering if it will work properly.
[image]

actuallymentor, 2023-02-04:
I just pushed a update that should fix a few issues, could you try to run `brew reinstall --cask battery` and see if that fixes it?

tylermilner, 2023-02-04:
I appreciate you looking into this. Unfortunately, it looks like v1.0.5 is giving the same behavior - 2 error messages before the app shows up in the menu bar.

[image]
[image]

actuallymentor, 2023-02-05:
Could you try the following things to help me debug:

**Debug trail 1: installing CLI only**

Instead of installing through brew, can you install through [these CLI instructions](https://github.com/actuallymentor/battery#installation-1) and show me your terminal output?

**Debug trail 2: running the app locally**

Can you try:

```shell
git clone git@github.com:actuallymentor/battery.git
cd battery/app
npm i
npm start
```

And show me what happens?

tylermilner, 2023-02-07:
As far as I can tell, the CLI installation appears to have worked. Here's the terminal output for the process of installation, enable, disable, and removal:

```Shell
tylermilner@Tylers-MBP ~ % curl -s https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | bash

####################################################################
# 👋 Welcome, this is the setup script for the battery CLI tool.
# Note: this script will ask for your password once or multiple times.
####################################################################

Password:
🔋 Starting battery installation

[ 1/10 ] Xcode build tools are installed, continuing
[ 2/10 ] Superuser permissions acquired.
[ 3/10 ] Cloning fan control version of smc
[ 4/10 ] Building smc from source
[ 5/10 ] Move smc to executable folder
[ 6/10 ] Cloning battery repository
[ 7/10 ] Writing script to /usr/local/bin/battery for user tylermilner
[ 8/10 ] Set up visudo declarations
[ 9/10 ] Removing temp folder /Users/tylermilner/.battery-tmp
[ 10/10 ] Removed temporary build files

🎉 Battery tool installed. Type "battery help" for instructions.

tylermilner@Tylers-MBP ~ % battery status
02/05/23-19:18:39 - Battery at 81% (8:27 remaining), smc charging enabled
02/05/23-19:18:39 - Your battery is currently being maintained at %
tylermilner@Tylers-MBP ~ % battery maintain 80
02/05/23-19:20:50 - Starting battery maintenance at 80%
02/05/23-19:20:50 - Maintaining battery at 80%
tylermilner@Tylers-MBP ~ % battery maintain stop
02/05/23-19:22:06 - Killing running maintain daemons & enabling charging as default state
02/05/23-19:22:06 - 🔌🔋 Enabling battery charging
02/05/23-19:22:06 - 🔼🪫 Disabling battery discharging
02/05/23-19:22:06 - Battery at 81% (8:57 remaining), smc charging enabled
tylermilner@Tylers-MBP ~ % battery uninstall
This will enable charging, and remove the smc tool and battery script
Press any key to continue

02/05/23-19:22:22 - 🔌🔋 Enabling battery charging
02/05/23-19:22:22 - 🔼🪫 Disabling battery discharging
02/05/23-19:22:22 - 🔼🪫 Disabling battery discharging
/usr/local/bin/smc
/usr/local/bin/battery
```

Running the app locally via `npm` was not successful. There were 4 error popups:

[image]

[image]

[image]

[image]

After dismissing the errors, the battery icon was visible in the menu bar, but it was unresponsive. I ended up killing the process with `CONTROL + C`. Here's the full terminal output for git clone, npm install, and run:

```Shell
tylermilner@Tylers-MBP Repositories % git clone git@github.com:actuallymentor/battery.git
Cloning into 'battery'...
Enter passphrase for key '/Users/tylermilner/.ssh/id_ed25519': 
remote: Enumerating objects: 506, done.
remote: Counting objects: 100% (229/229), done.
remote: Compressing objects: 100% (120/120), done.
remote: Total 506 (delta 133), reused 126 (delta 108), pack-reused 277
Receiving objects: 100% (506/506), 484.88 KiB | 1.75 MiB/s, done.
Resolving deltas: 100% (290/290), done.
[... 91 lines cut ...]
Error getting battery status:  Error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

    at ChildProcess.exithandler (node:child_process:407:12)
    at ChildProcess.emit (node:events:527:28)
    at maybeClose (node:internal/child_process:1092:16)
    at Socket.<anonymous> (node:internal/child_process:451:11)
    at Socket.emit (node:events:527:28)
    at Pipe.<anonymous> (node:net:709:12) {
  code: 127,
  killed: false,
  signal: null,
  cmd: 'PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv'
}
Error generating menu:  TypeError: (intermediate value) is not iterable
    at generate_app_menu (/Users/tylermilner/Repositories/battery/app/modules/interface.js:16:73)
    at async refresh_tray (/Users/tylermilner/Repositories/battery/app/modules/interface.js:95:22)
    at async set_initial_interface (/Users/tylermilner/Repositories/battery/app/modules/interface.js:139:5)
(node:32522) UnhandledPromiseRejectionWarning: TypeError: Must pass Menu or null
    at refresh_tray (/Users/tylermilner/Repositories/battery/app/modules/interface.js:101:10)
    at async set_initial_interface (/Users/tylermilner/Repositories/battery/app/modules/interface.js:139:5)
(Use `Electron --trace-warnings ...` to show where the warning was created)
(node:32522) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
^C/Users/tylermilner/Repositories/battery/app/node_modules/electron/dist/Electron.app/Contents/MacOS/Electron exited with signal SIGINT
```

dyipon, 2023-02-07:
just mention, this app is working on a macbook air m2, for me.

actuallymentor, 2023-02-07:
@tylermilner can you `git pull` and try again? I think this is the bug that was closed in the latest push.

tylermilner, 2023-02-07:
Looks like it's getting closer, but there are still some issues.

[image]
[image]
[image]
[image]

Terminal output:

```Shell
tylermilner@Tylers-MBP battery % git status
On branch main
Your branch is behind 'origin/main' by 7 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

nothing to commit, working tree clean
tylermilner@Tylers-MBP battery % git pull 
Enter passphrase for key '/Users/tylermilner/.ssh/id_ed25519': 
Updating 6f6d4eb..ffc34cd
Fast-forward
 README.md                                         |   2 +-
 app/assets/battery-active-20.png                  | Bin 0 -> 682 bytes
 app/assets/battery-active-20@1.25x.png            | Bin 0 -> 753 bytes
 app/assets/battery-active-20@1.33x.png            | Bin 0 -> 744 bytes
 app/assets/battery-active-20@1.4x.png             | Bin 0 -> 761 bytes
 app/assets/battery-active-20@1.5x.png             | Bin 0 -> 775 bytes
 app/assets/battery-active-20@1.8x.png             | Bin 0 -> 808 bytes
 app/assets/battery-active-20@2.5x.png             | Bin 0 -> 978 bytes
 app/assets/battery-active-20@2x.png               | Bin 0 -> 866 bytes
 app/assets/battery-active-20@3x.png               | Bin 0 -> 1069 bytes
 app/assets/battery-active-20@4x.png               | Bin 0 -> 1271 bytes
 app/assets/battery-active-20@5x.png               | Bin 0 -> 1486 bytes
 app/assets/battery-active-50.png                  | Bin 0 -> 696 bytes
 app/assets/battery-active-50@1.25x.png            | Bin 0 -> 773 bytes
 app/assets/battery-active-50@1.33x.png            | Bin 0 -> 782 bytes
 app/assets/battery-active-50@1.4x.png             | Bin 0 -> 782 bytes
 app/assets/battery-active-50@1.5x.png             | Bin 0 -> 794 bytes
 app/assets/battery-active-50@1.8x.png             | Bin 0 -> 863 bytes
 app/assets/battery-active-50@2.5x.png             | Bin 0 -> 1019 bytes
 app/assets/battery-active-50@2x.png               | Bin 0 -> 913 bytes
 app/assets/battery-active-50@3x.png               | Bin 0 -> 1130 bytes
 app/assets/battery-active-50@4x.png               | Bin 0 -> 1365 bytes
 app/assets/battery-active-50@5x.png               | Bin 0 -> 1602 bytes
 app/assets/battery-active-80.png                  | Bin 0 -> 684 bytes
 app/assets/battery-active-80@1.25x.png            | Bin 0 -> 774 bytes
 app/assets/battery-active-80@1.33x.png            | Bin 0 -> 772 bytes
 app/assets/battery-active-80@1.4x.png             | Bin 0 -> 785 bytes
 app/assets/battery-active-80@1.5x.png             | Bin 0 -> 790 bytes
 app/assets/battery-active-80@1.8x.png             | Bin 0 -> 865 bytes
 app/assets/battery-active-80@2.5x.png             | Bin 0 -> 1024 bytes
 app/assets/battery-active-80@2x.png               | Bin 0 -> 903 bytes
 app/assets/battery-active-80@3x.png               | Bin 0 -> 1133 bytes
 app/assets/battery-active-80@4x.png               | Bin 0 -> 1360 bytes
 app/assets/battery-active-80@5x.png               | Bin 0 -> 1600 bytes
 app/assets/battery-active-darkmode-20.png         | Bin 0 -> 696 bytes
 app/assets/battery-active-darkmode-20@1.25x.png   | Bin 0 -> 782 bytes
 app/assets/battery-active-darkmode-20@1.33x.png   | Bin 0 -> 775 bytes
 app/assets/battery-active-darkmode-20@1.4x.png    | Bin 0 -> 796 bytes
 app/assets/battery-active-darkmode-20@1.5x.png    | Bin 0 -> 810 bytes
 app/assets/battery-active-darkmode-20@1.8x.png    | Bin 0 -> 849 bytes
 app/assets/battery-active-darkmode-20@2.5x.png    | Bin 0 -> 1025 bytes
 app/assets/battery-active-darkmode-20@2x.png      | Bin 0 -> 902 bytes
 app/assets/battery-active-darkmode-20@3x.png      | Bin 0 -> 1126 bytes
 app/assets/battery-active-darkmode-20@4x.png      | Bin 0 -> 1328 bytes
 app/assets/battery-active-darkmode-20@5x.png      | Bin 0 -> 1577 bytes
 app/assets/battery-active-darkmode-50.png         | Bin 0 -> 717 bytes
 app/assets/battery-active-darkmode-50@1.25x.png   | Bin 0 -> 800 bytes
 app/assets/battery-active-darkmode-50@1.33x.png   | Bin 0 -> 821 bytes
 app/assets/battery-active-darkmode-50@1.4x.png    | Bin 0 -> 811 bytes
 app/assets/battery-active-darkmode-50@1.5x.png    | Bin 0 -> 829 bytes
[... 304 lines cut ...]
Error getting battery status:  Error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

    at ChildProcess.exithandler (node:child_process:407:12)
    at ChildProcess.emit (node:events:527:28)
    at maybeClose (node:internal/child_process:1092:16)
    at ChildProcess._handle.onexit (node:internal/child_process:302:5) {
  code: 127,
  killed: false,
  signal: null,
  cmd: 'PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv'
}
Error generating menu:  TypeError: Cannot destructure property 'battery_state' of '(intermediate value)' as it is undefined.
    at generate_app_menu (/Users/tylermilner/Repositories/battery/app/modules/interface.js:16:17)
    at process.processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async refresh_tray (/Users/tylermilner/Repositories/battery/app/modules/interface.js:96:22)
    at async set_initial_interface (/Users/tylermilner/Repositories/battery/app/modules/interface.js:159:5)
(node:4139) UnhandledPromiseRejectionWarning: TypeError: Must pass Menu or null
    at refresh_tray (/Users/tylermilner/Repositories/battery/app/modules/interface.js:102:10)
    at process.processTicksAndRejections (node:internal/process/task_queues:96:5)
[... cut ...]
(node:4139) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
^C/Users/tylermilner/Repositories/battery/app/node_modules/electron/dist/Electron.app/Contents/MacOS/Electron exited with signal SIGINT
```

actuallymentor, 2023-02-07:
Can you add `tail -n 100 ~/.battery/battery.log` to this? It looks like the installation fails for some reason.

Just to be sure: you get a popup asking for your admin password right?

tylermilner, 2023-02-08:
Yes, I do get the initial popup, but no password prompting happens after clicking "OK". Instead it goes straight to the 4 error popups.
[image]

Here is the output of `tail -n 100 ~/.battery/battery.log`:

```Shell
tylermilner@Tylers-MBP app % tail -n 100 ~/.battery/battery.log
02/05/23-19:20:50 - Triggering discharge to 80 before enabling charging limiter
02/05/23-19:20:50 - Discharging to 80% from 81%
02/05/23-19:20:50 - 🔽🪫 Enabling battery discharging
02/05/23-19:20:50 - Battery at 81% (target 80%)
02/05/23-19:21:50 - Battery at 81% (target 80%)
```

actuallymentor, 2023-02-08:
Ah interesting, that would explain the issue. The first thing the app does is install the under-the-hook utility with `sudo` rights, for which it should prompt.

I added some more verbose logging, can you `git pull && npm i` again?

Than you for taking the time to walk through this @tylermilner!

tylermilner, 2023-02-08:
Here's the latest run:
[image]
[image]
[image]
[image]
[image]

```Shell
tylermilner@Tylers-MBP app % git pull
Updating ffc34cd..b372c8f
Fast-forward
 README.md                                         |   2 +-
 app/assets/battery-active-20.png                  | Bin 682 -> 682 bytes
 app/assets/battery-active-20@1.25x.png            | Bin 753 -> 753 bytes
 app/assets/battery-active-20@1.33x.png            | Bin 744 -> 744 bytes
 app/assets/battery-active-20@1.4x.png             | Bin 761 -> 761 bytes
 app/assets/battery-active-20@1.5x.png             | Bin 775 -> 775 bytes
 app/assets/battery-active-20@1.8x.png             | Bin 808 -> 808 bytes
 app/assets/battery-active-20@2.5x.png             | Bin 978 -> 978 bytes
 app/assets/battery-active-20@2x.png               | Bin 866 -> 866 bytes
 app/assets/battery-active-20@3x.png               | Bin 1069 -> 1069 bytes
 app/assets/battery-active-20@4x.png               | Bin 1271 -> 1271 bytes
 app/assets/battery-active-20@5x.png               | Bin 1486 -> 1486 bytes
 app/assets/battery-active-50.png                  | Bin 696 -> 696 bytes
 app/assets/battery-active-50@1.25x.png            | Bin 773 -> 773 bytes
 app/assets/battery-active-50@1.33x.png            | Bin 782 -> 782 bytes
 app/assets/battery-active-50@1.4x.png             | Bin 782 -> 782 bytes
 app/assets/battery-active-50@1.5x.png             | Bin 794 -> 794 bytes
 app/assets/battery-active-50@1.8x.png             | Bin 863 -> 863 bytes
 app/assets/battery-active-50@2.5x.png             | Bin 1019 -> 1019 bytes
 app/assets/battery-active-50@2x.png               | Bin 913 -> 913 bytes
 app/assets/battery-active-50@3x.png               | Bin 1130 -> 1130 bytes
 app/assets/battery-active-50@4x.png               | Bin 1365 -> 1365 bytes
 app/assets/battery-active-50@5x.png               | Bin 1602 -> 1602 bytes
 app/assets/battery-active-80.png                  | Bin 684 -> 684 bytes
 app/assets/battery-active-80@1.25x.png            | Bin 774 -> 774 bytes
 app/assets/battery-active-80@1.33x.png            | Bin 772 -> 772 bytes
 app/assets/battery-active-80@1.4x.png             | Bin 785 -> 785 bytes
 app/assets/battery-active-80@1.5x.png             | Bin 790 -> 790 bytes
 app/assets/battery-active-80@1.8x.png             | Bin 865 -> 865 bytes
 app/assets/battery-active-80@2.5x.png             | Bin 1024 -> 1024 bytes
 app/assets/battery-active-80@2x.png               | Bin 903 -> 903 bytes
 app/assets/battery-active-80@3x.png               | Bin 1133 -> 1133 bytes
 app/assets/battery-active-80@4x.png               | Bin 1360 -> 1360 bytes
 app/assets/battery-active-80@5x.png               | Bin 1600 -> 1600 bytes
 app/assets/battery-active-95.png                  | Bin 0 -> 683 bytes
 app/assets/battery-active-95@1.25x.png            | Bin 0 -> 763 bytes
 app/assets/battery-active-95@1.33x.png            | Bin 0 -> 768 bytes
 app/assets/battery-active-95@1.4x.png             | Bin 0 -> 765 bytes
 app/assets/battery-active-95@1.5x.png             | Bin 0 -> 789 bytes
 app/assets/battery-active-95@1.8x.png             | Bin 0 -> 846 bytes
 app/assets/battery-active-95@2.5x.png             | Bin 0 -> 1009 bytes
 app/assets/battery-active-95@2x.png               | Bin 0 -> 890 bytes
 app/assets/battery-active-95@3x.png               | Bin 0 -> 1129 bytes
 app/assets/battery-active-95@4x.png               | Bin 0 -> 1331 bytes
 app/assets/battery-active-95@5x.png               | Bin 0 -> 1602 bytes
 app/assets/battery-active-darkmode-20.png         | Bin 696 -> 696 bytes
 app/assets/battery-active-darkmode-20@1.25x.png   | Bin 782 -> 782 bytes
 app/assets/battery-active-darkmode-20@1.33x.png   | Bin 775 -> 775 bytes
 app/assets/battery-active-darkmode-20@1.4x.png    | Bin 796 -> 796 bytes
 app/assets/battery-active-darkmode-20@1.5x.png    | Bin 810 -> 810 bytes
 app/assets/battery-active-darkmode-20@1.8x.png    | Bin 849 -> 849 bytes
 app/assets/battery-active-darkmode-20@2.5x.png    | Bin 1025 -> 1025 bytes
 app/assets/battery-active-darkmode-20@2x.png      | Bin 902 -> 902 bytes
 app/assets/battery-active-darkmode-20@3x.png      | Bin 1126 -> 1126 bytes
 app/assets/battery-active-darkmode-20@4x.png      | Bin 1328 -> 1328 bytes
 app/assets/battery-active-darkmode-20@5x.png      | Bin 1577 -> 1577 bytes
 app/assets/battery-active-darkmode-50.png         | Bin 717 -> 717 bytes
 app/assets/battery-active-darkmode-50@1.25x.png   | Bin 800 -> 800 bytes
 app/assets/battery-active-darkmode-50@1.33x.png   | Bin 821 -> 821 bytes
[... 224 lines cut ...]
Error getting battery status:  Error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

    at ChildProcess.exithandler (node:child_process:407:12)
    at ChildProcess.emit (node:events:527:28)
    at maybeClose (node:internal/child_process:1092:16)
    at ChildProcess._handle.onexit (node:internal/child_process:302:5) {
  code: 127,
  killed: false,
  signal: null,
  cmd: 'PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv'
}
Error generating menu:  TypeError: Cannot destructure property 'battery_state' of '(intermediate value)' as it is undefined.
    at generate_app_menu (/Users/tylermilner/Repositories/battery/app/modules/interface.js:16:17)
    at process.processTicksAndRejections (node:internal/process/task_queues:96:5)
[... cut ...]
(node:17798) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
^C/Users/tylermilner/Repositories/battery/app/node_modules/electron/dist/Electron.app/Contents/MacOS/Electron exited with signal SIGINT
```

actuallymentor, 2023-02-08:
Oof I shamefully forgot to push my local changes, could you `git pull && npm i && npm start` again? I need only the logs.

tylermilner, 2023-02-08:
Sure, no problem. Here is the `gui.log` after running `git pull`, `npm i`, and `npm start` on commit `172b53e` (there was no `battery.log` file):

```
Starting tray app
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ curl github.com &> /dev/null
Display percentage 95 based on 100
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ curl icanhasip.com &> /dev/null
Triggering boot-time auto-update
Get inactive logo for 100
Tray app boot complete
Internet online: true
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ which git
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ sudo -n /usr/local/bin/smc -k CH0C -r
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ which battery
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ which smc
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ sudo -n /usr/local/bin/smc -k CH0I -r
Is installed? 
false
Executing ps aux | grep "/usr/local/bin/battery " | wc -l | grep -Eo "\d*"
Found 2 battery related processed to kill
Executing pkill -f "/usr/local/bin/battery.*"
Error killing existing battery progesses, usually means no running processes
Installing battery for tylermilner...
Sudo executing command: curl -s https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | bash -s -- $USER
Update/install error: 
Error: Command failed: ./applet
/bin/sh: ./applet: Bad CPU type in executable

App initialisation process complete
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Error getting battery status: 
Error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery maintain 80
Error enabling battery: 
Error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery maintain 80
/bin/bash: battery: command not found

Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Refreshing tray icon...
Error getting battery status: 
Error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

Error generating menu: 
TypeError: Cannot destructure property 'battery_state' of '(intermediate value)' as it is undefined.
```

actuallymentor, 2023-02-11:
Hmm ok I did some sleuthing, and there is a chance this relates to the `electron-sudo` module I'm using. The [sudo applet for the OSX part of](https://github.com/automation-stack/electron-sudo/tree/master/src/bin/applet.app) the package is 7 years old, and it is the only reference to the word 'applet' in the entire codebase and included dependency tree.

My guess: that applet is not arm compatible.

What confuses me: some people get this `/bin/sh: ./applet: Bad CPU type in executable` error but some (like me) never do...

Can you perhaps try to see if you have rosetta (the `amd64` sandbox for `arm` mac devices) installed:

```shell
pgrep oahd
lsbom -f /Library/Apple/System/Library/Receipts/com.apple.pkg.RosettaUpdateAuto.bom
```

If you see no output above, please let me know and then try to run `/usr/sbin/softwareupdate --install-rosetta --agree-to-license`

actuallymentor, 2023-02-11:
Actually, scratch that. Can you git pull and try again?

I can't believe I'm saying this, but I think I worked around it by having the electron app trigger applescript which triggers the shell script that triggers the installation.

It worked on my local machine, and it it does on yours then we have bypassed the need for fixing the `electron-sudo` package.

tylermilner, 2023-02-12:
@actuallymentor I think you've done it! As you suspected, this machine doesn't have Rosetta since it's still pretty fresh out of the box without having installed anything that requires it yet:

```Shell
tylermilner@Tylers-MBP app % pgrep oahd
tylermilner@Tylers-MBP app % lsbom -f /Library/Apple/System/Library/Receipts/com.apple.pkg.RosettaUpdateAuto.bom
can't open /Library/Apple/System/Library/Receipts/com.apple.pkg.RosettaUpdateAuto.bom: No such file or directory
**** Can't open /Library/Apple/System/Library/Receipts/com.apple.pkg.RosettaUpdateAuto.bom.
```

After doing a `git pull && npm i && npm start`, I got the welcome popup and then used Touch ID to install the helper script. Everything seems to be working now. 🎉 Thanks for your hard work on this!

[image]

[image]

[image]

[image]

closed 2023-02-12 by tylermilner

actuallymentor, 2023-02-12:
Awesome, I'll do a refactor and deploy a `v1.0.8`

actuallymentor, 2023-02-12:
[Pre release here](https://github.com/actuallymentor/battery/releases/tag/v1.0.8-beta) if you want to test it

tylermilner, 2023-02-12:
@actuallymentor The pre-release looks good! I checked out the `v1.0.8-beta` tag and everything is working.