actuallymentor/battery #49: Macbook pro M1 // stuck at updating...

71m383nd3r, 2023-02-02:
**stuck at menubar with message updating....**
[image]

more then 20min...

_Macbook Pro 2020 M1, MacOS **Monteray** v12.6.3_
**Battery v1.0.4**

actuallymentor, 2023-02-04:
I just pushed a update that should fix a few issues, could you try to run `brew reinstall --cask battery` and see if that fixes it?

71m383nd3r, 2023-02-04:
[image]

>` brew reinstall --cask battery`

[image]

some other suggestion?
tnx 

I tried as well v1.0.5  - same problem.

actuallymentor, 2023-02-04:
Ah yes that is a [brew issue](https://github.com/Homebrew/homebrew-cask/issues/140411) that they are trying to fix. Nothing we can do about that for now I'm afraid. Try it again tomorrow.

JacksonChen666, 2023-02-05:
> Ah yes that is a [brew issue](https://github.com/Homebrew/brew/issues/14504) that they are trying to fix. Nothing we can do about that for now I'm afraid. Try it again tomorrow.

referenced issue is closed, @71m383nd3r try again with `brew reinstall --force battery` (https://github.com/Homebrew/brew/issues/14504#issuecomment-1417747268)

closed 2023-02-05 by actuallymentor

71m383nd3r, 2023-02-05:
[image]
I tried;

 `brew reinstall --force battery`

` brew install --force battery`

still the same problem: `Download failed on Cask 'battery' with message: Download failed: https://github.com/actuallymentor/battery/releases/download/v1.0.5/battery-1.0.5-arm64.dmg`