actuallymentor/battery #50: Mac keeps asking for sudo password after uninstalling the utility

boof-fo, 2023-02-05:
**What is the issue? (required)**
My mac keeps asking for sudo password even after I uninstalled the app. I know it's asking the password because of Battery since I've opened Activity Monitor and I checked which files were open in the bash process that shows the sudo password prompt. It shows "/usr/local/bin/battery" but the file has been removed.

**What exactly did you do to produce the issue? (required)**
I uninstalled it through the command "battery uninstall". It didn't work so I tried reinstalling it and uninstalling it through brew. Didn't work either.

actuallymentor, 2023-02-05:
Can you try running: `pkill -f "/usr/local/bin/battery.*"`?

boof-fo, 2023-02-06:
Thank you, it worked

actuallymentor, 2023-02-07:
Thanks for confirming. I added an auto-kill to the uninstall script, so it should not happen again.

closed 2023-02-07 by actuallymentor