actuallymentor/battery #51: Impossible to install the dmg on M1 pro 14"

LucasMSalgado, 2023-02-05:
**What is the issue? (required)**
Impossible to install the battery software using the option with the dmg file.

**What exactly did you do to produce the issue? (required)**
After copying the dmg to the application folder,
Launching the dmg for the installation leads to multiples errors :
[image]
[image]
[image]
[image]

Then the installation finished and the icon is on the top bar and on the application list, but it is impossible to launch it or setup it.
[image]

**Expected behavior (required)**
I expected has explained in the tutorial to be able to acces to the menu on the top bar and be able to launch the application.

**Additional context (optional)**
Thanks you a lot for your soft, the dmg file make it very more user friendly.

Respectfully,
Lucas

st-ivan, 2023-02-05:
hi, just updated to 1.06, same issues here. 
1.05 works fine.. no discharge functionality tho.

NdR91, 2023-02-05:
I was creating the same issue, then I saw this one.

Same errors with this dmg

st-ivan, 2023-02-05:
ok, i uninstalled 1.06, reainstalled 1.05, disabled limit, uninstalled 1.05, installed and reinstalled 1.06 and it now works. but im not getting same info as in the scrennshot.. heres what i got:

[image]

NdR91, 2023-02-05:
Ok I found a similar workaround. 

Disabling the battery limit, you can quit the application and overwrite the 1.0.6 release on the 1.0.5. Prior to this I updated the CLI version.

> ok, i uninstalled 1.06, reainstalled 1.05, disabled limit, uninstalled 1.05, installed and reinstalled 1.06 and it now works. but im not getting same info as in the scrennshot.. heres what i got:
> 
> [image]
> 
> m1pro 14 im going back to 1.05 until 1.06 is fixed.

Regarding this, do you also set the `battery discharge` setting?
If not it won't discharge if you plug the power over that percentage.

actuallymentor, 2023-02-05:
@LucasMSalgado tjabks for letting me know. Can you show me the output of `battery logs`?

actuallymentor, 2023-02-05:
Found the issue. Foolish me. With I knew more about automated testing for mac apps...

Will push a fix within a few hours

actuallymentor, 2023-02-05:
Ok, so because I am a terrible person I have replaced the release dmg with the fixed one without incrementing the version number. Can you download it and verify if works for you?

NdR91, 2023-02-05:
> Ok, so because I am a terrible person I have replaced the release dmg with the fixed one without incrementing the version number. Can you download it and verify if works for you?

Installation went well with this dmg.

FYI: the update overwrites the preferences.
My "maintain" option is set on 75%.

After the update:
```
╭─ndr@NdRs-MBP.local /  
╰─➤  battery status
02/05/23-20:08:15 - Battery at 76% (attached; remaining), smc charging enabled
02/05/23-20:08:15 - Your battery is currently being maintained at 80%
```

[image]

Note a but issue, but maybe you want to know it.

actuallymentor, 2023-02-05:
The GUI only supports 80%, though in this new version if you run the cli at 75, the GUI will show 75 (but will default back to 80 when you disable).

The GUI is mostly for normies, the cli for custom stuff.

closed 2023-02-05 by actuallymentor