actuallymentor/battery #52: Homebrew Version 1.0.5 doesn't update to 1.0.6

NdR91, 2023-02-05:
**What is the issue? (required)**
Homebrew installation doesn't find the new version (1.0.6)

**What exactly did you do to produce the issue? (required)**
With the `battery update` command it keeps the 1.0.5 version

```
This will run curl -sS https://raw.githubusercontent.com/actuallymentor/battery/main/update.sh | bash
Press any key to continue

🔋 Starting battery update

[ 1/3 ] Cloning battery repository
[ 2/3 ] Writing script to /usr/local/bin/battery
[ 3/3 ] Removed temporary folder

🎉 Battery tool updated.
```

**Screenshots (optional)**
[image]
**Error logs**
If you know how a terminal works, please paste the content of `battery logs` here. This allows me to see potential errors on your device.

NdR91, 2023-02-05:
I just found out that homebrew updates doesn't include the GUI version. So I sill have to install it via DMG, right?

actuallymentor, 2023-02-05:
To update the home brew version takes a day or so. Whenever I update the app I need to open a pull request with them, and they are very diligent about checking them.

closed 2023-02-05 by actuallymentor

NdR91, 2023-02-05:
> To update the home brew version takes a day or so. Whenever I update the app I need to open a pull request with them, and they are very diligent about checking them.

Oh ok! Thanks for the feedback