actuallymentor/battery #56: Menu bar icon suddenly white

kristiankruse, 2023-02-09:
After installing 1.0.6 and also upgraded to 1.0.7 the menu bar icon is now white—whereas it used to be black to match the rest of the icons.

schabrolles, 2023-02-11:
I had the opposite behaviour ... install 1.0.7, try icon is black while it should be white

actuallymentor, 2023-02-11:
Hmm it should keep to your light/dark system setting. Screenshot?

schabrolles, 2023-02-12:
Here it is ... I'm using macOS Ventura .... this mean that light mode does not mean dark tray icon ... It depends of your wallpaper background. I'm with "auto" light/dark mode in macOS. My background is more dark than light. So even during the day (light mode) tray icon must be white.

Here is a screenshot during the day, light them in macOS, all the tray icons are white except "battery.app"
[image]

abstractrealism, 2023-02-13:
I'm experiencing the same on Monterrey. Not sure if it's because of background color or what, but all icons besides battery are white during the day. Has been the case since I installed it a while ago, not a new issue. I thought it was intentional haha

kristiankruse, 2023-02-14:
[image]

Neifab, 2023-02-15:
MacOS Ventura 13.2.1, version 1.0.8

[image]

nomadplanet, 2023-02-27:
The menu bar does not necessarily update to a white theme when switching to dark mode or a black theme when switching off the dark mode.  It's all about the wallpaper colour behind it. For instance, my menu bar icons are always white whether I'm in dark or light mode. 

The problem is that battery follows entirely the dark/light mode to decide which icon variant to apply.

Neifab, 2023-02-27:
> 

Yes, should be easy to fix

earthsound, 2023-03-01:
@actuallymentor this article describes how to set up [macOS monochrome template images](https://bjango.com/articles/designingmenubarextras/#colour-and-template-images) to resolve this.

actuallymentor, 2023-03-06:
Should be fixed in latest release [v1.1.1](https://github.com/actuallymentor/battery/releases/tag/v1.1.1)

closed 2023-03-06 by actuallymentor

nomadplanet, 2023-03-06:
Indeed, it is 👍 
Thanks