actuallymentor/battery #57: Low resolution menu bar icon

peterneutron, 2023-02-11:
**What is the issue? (required)**
Low resolution menu bar icon being displayed.

**What exactly did you do to produce the issue? (required)**
Run `battery 1.0.7`

**Expected behavior (required)**
High resolution menu bar icon being displayed.

**Screenshots (optional)**
[image]

**Additional context (optional)**
`M1 MBP 13"` `macOS 13.2 (22D49)` `battery 1.0.7`

actuallymentor, 2023-02-11:
Would this happen to be on an external monitor?

peterneutron, 2023-02-11:
Nope, internal display.

actuallymentor, 2023-02-12:
Very odd, I'm unable to reproduce. Does anything change when you click the tray icon?

Could you show me:

1. A screenshot
2. The output of `battery logs`

peterneutron, 2023-02-12:
@actuallymentor You're missing all `battery-active-darkmode-95@` files in the `asset` directory. There is only a single one `battery-active-darkmode-95.png`. I should have checked for these way earlier 😅. 

As for the icon missing below 20% in dark mode, I'll have a look later maybe that one was just a fluke.

actuallymentor, 2023-02-12:
Ah well spotted! I'll add it in the official release of `v1.0.8`

peterneutron, 2023-02-12:
Okay, I checked again and the icon definitely goes missing below 20% in both light and dark mode.

https://user-images.githubusercontent.com/60703175/218325379-6178d0d1-1c16-4444-8689-e5b1daab4f13.mp4

closed 2023-02-13 by actuallymentor

peterneutron, 2023-02-13:
@actuallymentor Missing icon under 20% still isn't fixed in `1.0.8`

actuallymentor, 2023-02-13:
@peterneutron which image are you missing in [this dir](https://github.com/actuallymentor/battery/tree/main/app/assets)?

mn113, 2023-02-13:
In case it's related, I have the transparent menubar on Ventura/13.2.0, and `battery` renders with a black icon:
[image]

(white Dropbox for comparison)
Even though I can see all the white and black copies of the icon are installed in `battery.app/Contents/Resources`.

The icon also randomly disappears. I have v1.0.8 installed.

peterneutron, 2023-02-13:
@actuallymentor Did you look at my posted video above? As soon as battery level drops to =< 20% the whole icon goes missing. Also diffing the assest from the repo with the assets on my local installation comes up with no missing files.

earthsound, 2023-02-17:
When < 20% the following error shows up in the log:
```
Display percentage undefined based on 19
```

Once it gets to 20% or above, it changes to this:
```
Display percentage 20 based on 21
```
I have to interact with the (blank) menu bar icon/space when the battery gets at/above 20% and then the icon shows up.

earthsound, 2023-03-06:
@actuallymentor was the issue w/ the icon disappearing when the battery is <20% also fixed in v1.1.1?

earthsound, 2023-03-06:
@actuallymentor the disappearing icon <20% issue is still present. Can this issue be reopened/renamed?
[image]

earthsound, 2023-03-06:
@actuallymentor relevant lines from battery log:
```
Get active logo for 18
Display percentage undefined based on 18
🚨 Missing image: /Applications/Battery.app/Contents/Resources/battery-active-undefinedTemplate.png
```

earthsound, 2023-03-07:
@actuallymentor from a cursory glance, it appears (at least) two things are needed:

1. A set of template icons to cover battery percentages from 0-19 (e.g., battery-active-0Template.png, etc.)
2. Add 0 in line 18 of theme.js at the beginning of the const array:
```
    const image_percentages = [ 0, 20, 50, 60, 70, 80, 95 ].sort()
```
I didn't look deeply to determine whether these types of additions need to be added elsewhere.

actuallymentor, 2023-03-07:
@earthsound thanks for the details. I'll do a quick fox next release by including:

```js
let display_percentage = 20
```

as a default.

The extra template images is something I'm looking into. I'm having trouble getting Electron to accept `svg`s as Tray image, which would allow me to programatically change the battery fullness.

actuallymentor, 2023-03-07:
Turns out I'm not crazy and electron simply does not support `svg` for native images [electron/issues/9642](https://github.com/electron/electron/issues/9642). I'll look into a workaround.

actuallymentor, 2023-03-07:
@earthsound give this a whack? https://github.com/actuallymentor/battery/releases/tag/v1.1.2-beta

earthsound, 2023-03-07:
@actuallymentor 1.1.2 is working as expected
[image]

commit 4a387ba6 (2023-02-13): 🐛 missing darkmode icon, closes #57