actuallymentor/battery #58: Add battery percentage in de bar near the icon

jcbernard83, 2023-02-11:
Add battery percentage in de bar near the icon

schabrolles, 2023-02-11:
Having more customisation like show the time remaining and/or battery %

actuallymentor, 2023-02-11:
I'm a bit torn on this. I kind of want to keep this tool very minimalist, especially since many of the commonly asked features are built into the battery icon that the system already has.

Can you explain why you want this feature? Why not just use the built-in battery icon for this?

jcbernard83, 2023-02-12:
Hi,I want to avoid 2 battery icon in the bar. If battery diplays the percentage I can remove the OS icon. May be you can add too in the menu an item to access to the battery configuration. JC.

actuallymentor, 2023-02-12:
Is the current dropdown insufficient for you? It shows very similarly to the system dialogue:

[image]

[image]

schabrolles, 2023-02-12:
well ... I also would like to only have one battery icon... I really like having the possibility to get time remaining battery...

actuallymentor, 2023-02-12:
But it says the time remaining?

schabrolles, 2023-02-12:
No, the official battery tray icon gives you only battery percentage. While the command battery status can give you time remaining

actuallymentor, 2023-02-14:
It gives both, I'm very confused what you are talking about

jakjakob, 2023-02-15:
you can use just the Command-line version, so you'll have only the system icon