actuallymentor/battery #59: MacBook goes to sleep when external monitor is connected, only work with integrated (open) display

Ca-Wi, 2023-02-11:
MacBook goes to sleep when external monitor is connected, the battery app only work with integrated (open) display.
If I close the internal display the MB goes to sleep no wakeup possible.
If I start /resume the MacBook and I logged me in, "battery" starts the MB goes to sleep.

Apple M1 with MacOS 13.2 (22D49)

schabrolles, 2023-02-12:
@Ca-Wi try this ... it seems to work for me ...

without your screen ...open a terminal and enter :

```
battery adapter off && battery charge 80
```

This should keep the power to your Macbook when connected to your screen with thunderbolt but stop charging the battery if > 80%.
Note: the discharging function will be disabled .... but you can work ;)

artcrime, 2023-02-13:
I have the same problem even after uninstalling the app. Battery not charging anymore. Clamshell mode not working.
How can I switch to the settinge before I installed the app?

artcrime, 2023-02-13:
Removing from app folder did not work, reinstall / install via brew worked. The problem is that charging is disabled completely and if charing is disabled external displays are not working if in clamshell mode.

spkis, 2023-02-13:
Samne issue here.
While connected to a screen, using:
battery adapter off && battery charge 80

And closing the lid - the laptop is still working but also charging... I guess this needs a bit more tweaking.

schabrolles, 2023-02-13:
I think this is because of the "discharging mode" which is enabled by default when `battery maintain 80` is used...

I just disable it in the script `/usr/local/bin/battery` (by commenting line 348)

```
346         # Before we start maintaining the battery level, first discharge to the target level
347         log "Triggering discharge to $setting before enabling charging limiter"
348         #battery discharge "$setting"
349         log "Discharge pre battery-maintenance complete, continuing to battery maintenance loop"
```

Now everything seems to be OK for me ... I can use my external monitor with thunderbolt charging.
- MacBook is powered even when battery is >80% but battery is not charging 
- Battery is not discharging when connected to power and battery level >80%
- Battery is charging when when connected to power and battery level <80%

@actuallymentor, Do you think you can add a function to avoid battery discharging when plugged ? This should easily resolve that external monitor issue.

cholcombe973, 2023-02-16:
I also recently noticed this behavior but only after updating to the latest version of macos ventura (13.2.1). I don't know if my other macbook does this because it's not hooked up to an external monitor.

cholcombe973, 2023-02-27:
@schabrolles the part that might require some work is figuring out how to detect when the laptop is in clamshell mode. I found some old stack overflow stuff about it but I'm not sure how accurate it is. Another option would be maybe just exposing this as a user defined setting.

schabrolles, 2023-02-27:
You are right, detecting when the Mac is in clamshell mode is the solution for that issue

I think, proposing an option to also not discharge your laptop when plugged in could be interesting for the user …. It is also a good first step.

Cordialement / Best Regards
_________________________
Sébastien Chabrolles
Sr. technology Engineer
Red Hat & Hybrid Cloud on IBM Power<https://w3.ibm.com/w3publisher/ibm-systems-center-montpellier/red-hat-coc>
e-mail:     ***@***.******@***.***>

IBM Client Engineering | EMEA  <https://www.ibm.com/ibm/clientcenter/montpellier/index.shtml>
Montpellier

ZjQcmQRYFpfptBannerStart
This Message Is From an External Sender
This message came from outside your organization.

ZjQcmQRYFpfptBannerEnd

Unless otherwise stated above:

Compagnie IBM France
Siège Social : 17, avenue de l'Europe, 92275 Bois-Colombes Cedex
RCS Nanterre 552 118 465
Forme Sociale : S.A.S.
Capital Social : 664 069 390,60 €
SIRET : 552 118 465 03644 - Code NAF 6203Z

actuallymentor, 2023-03-08:
This issue is not pretty clear, it's a duplicate of #28, so I'm going to close this and refer anyone with this issue to that issue until I (or a contributor) finds some time to build this fix.

closed 2023-03-08 by actuallymentor