actuallymentor/battery #60: Installation fails both with .dmg and CLI

didotx, 2023-02-11:
**What is the issue? (required)**
**(1) install using .dmg**
downloaded the .dmg "battery-1.0.7-arm64.dmg" 
Install and run "battery tool"
fails: "Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv /bin/bash: battery: command not found"

**(2) reboot my Mac: did not fix the issue**

**(3) install using CLI**
from terminal:
> run " curl -s https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | bash"
result: 
- [ 1/10 ] Xcode build tools are installed, continuing
- [ 2/10 ] Superuser permissions acquired.
- [ 3/10 ] Cloning fan control version of smc
- bash: line 42: cd: /Users/didotx/.battery-tmp/smc/smc-command: No such file or directory
- [ 4/10 ] Building smc from source
- [ 5/10 ] Move smc to executable folder
- mv: rename /Users/didotx/.battery-tmp/smc/smc-command/smc to /usr/local/bin/smc: No such file or directory
- chmod: /usr/local/bin/smc: No such file or directory
- [ 6/10 ] Cloning battery repository
- [ 7/10 ] Writing script to /usr/local/bin/battery for user didotx
- cp: /Users/didotx/.battery-tmp/battery/battery.sh: No such file or directory
- chown: /usr/local/bin/battery: No such file or directory
- chmod: /usr/local/bin/battery: No such file or directory
- chmod: /usr/local/bin/battery: No such file or directory
- chown: /usr/local/bin/battery: No such file or directory
- bash: /Users/didotx/.battery-tmp/battery/battery.sh: No such file or directory
- [ 8/10 ] Set up visudo declarations
- [ 9/10 ] Removing temp folder /Users/didotx/.battery-tmp
- [ 10/10 ] Removed temporary build files
🎉 Battery tool installed. Type "battery help" for instructions.

> run: battery help
- bash: battery: command not found

> ls /usr/local/bin
espefuse.py		espsecure.py		esptool.py		pyserial-miniterm	pyserial-ports

>>> there is NO "/usr/local/bin/battery"
>>> Please advice

actuallymentor, 2023-02-11:
Could you show me the output of `which git curl`?

didotx, 2023-02-11:
Hi,
thank you so much for getting back to me. 
Here is the list of CLI commands and results:

Mecki-M1Pro:~ didotx$ curl -s https://raw.githubusercontent.com/actuallymentor/battery/main/setup.sh | bash

####################################################################
# 👋 Welcome, this is the setup script for the battery CLI tool.
# Note: this script will ask for your password once or multiple times.
####################################################################

🔋 Starting battery installation

[ 1/10 ] Xcode build tools are installed, continuing
[ 2/10 ] Superuser permissions acquired.
[ 3/10 ] Cloning fan control version of smc
bash: line 42: cd: /Users/didotx/.battery-tmp/smc/smc-command: No such file or directory
[ 4/10 ] Building smc from source
[ 5/10 ] Move smc to executable folder
mv: rename /Users/didotx/.battery-tmp/smc/smc-command/smc to /usr/local/bin/smc: No such file or directory
chmod: /usr/local/bin/smc: No such file or directory
[ 6/10 ] Cloning battery repository
[ 7/10 ] Writing script to /usr/local/bin/battery for user didotx
cp: /Users/didotx/.battery-tmp/battery/battery.sh: No such file or directory
chown: /usr/local/bin/battery: No such file or directory
chmod: /usr/local/bin/battery: No such file or directory
chmod: /usr/local/bin/battery: No such file or directory
chown: /usr/local/bin/battery: No such file or directory
bash: /Users/didotx/.battery-tmp/battery/battery.sh: No such file or directory
[ 8/10 ] Set up visudo declarations
[ 9/10 ] Removing temp folder /Users/didotx/.battery-tmp
[ 10/10 ] Removed temporary build files

🎉 Battery tool installed. Type "battery help" for instructions.

Mecki-M1Pro:~ didotx$ ls /usr/local/bin
espefuse.py		espsecure.py		esptool.py		pyserial-miniterm	pyserial-ports

Cheers, didotx

actuallymentor, 2023-02-12:
> Could you show me the output of `which git curl`?

Hey @didotx! I think you missed my above question

didotx, 2023-02-12:
Hi, I did send the info you asked for. 
here again.

actuallymentor, 2023-02-12:
Not what I asked for. Closing until you read.

closed 2023-02-12 by actuallymentor

didotx, 2023-02-12:
Hi, 
I am sorry, but I first did not get your question right. 
Please understand, I am quite new to github.

Trying again: https://github.com/actuallymentor/battery
I hope this is what you are looking for. 

cheers, didotx