actuallymentor/battery #61: Calibration mode

dalisoft, 2023-02-13:
**Is your feature request related to a problem? Please describe.**
Calibration mode to keep battery healthy, like AlDente feature exists but more automatized like exists `discharge` automatically for me did everything and i'm happy

**Describe the solution you'd like**
-

**Describe alternatives you've considered**
AlDente Calibration feature?

**Additional context**
OS: macOS 12.6.3 (Apple Silicon)

actuallymentor, 2023-02-14:
What do you mean with a calibration feature?

dalisoft, 2023-02-14:
@actuallymentor AlDente like Calibration mode. Like every two week automatically do steps: 100% charge, 20% discharge, then back to set value like 80%

actuallymentor, 2023-02-14:
I see. Are you asking for the GUI or CLI?

dalisoft, 2023-02-14:
@actuallymentor Any way. I can use CLI or GUI which is will be available first

rokk4, 2023-02-14:
Can't this be a simple cron job thing?

Check  last calibration timestamp , discharge if timstamp + 14 days < now then set semaphore -1
If sempahore -1, discharge if charge >= 20%
if charge < 20% set sempahore
If charge sempahore 1, charge until 100%
if charge is 100% set semaphore 0 and set new timestamp

This nice tool already has all the features needed, just need to glue them with some cron?
That's what I woud do under linux/bsd ... but I am quite new to macos, I don't know how lauchnd works yet :D

dalisoft, 2023-02-14:
I am not a new to macOS but still don't know how launchd works. 
i know, there AlDente but two tools with similar features will be cool and uses has a tool to choose. 

Maybe with a Account + licensing, i would pay as AlDente price but without limit to device count, only account or at least 5-6 devices per account for personal and up-to 7-10 devices for family plan :)

jpjenkins, 2023-03-19:
I have been attempting this but the issue https://github.com/actuallymentor/battery/issues/73#issue-1585174473 is causing problems. 
Here's my code so far to implement a calibration in aldente https://github.com/jpjenkins/actuallymentor-batteryCalibrationScript