actuallymentor/battery #64: install error (Air/M2 without rosetta installed)

henning-henemm, 2023-02-13:
**Battery isn't in the PATH**
I've used the latest .dmg (1.0.8)

Battery limiter error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery maintain 80
/bin/bash: battery: command not found

Battery limiter error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

**What exactly did you do to produce the issue? **
Installation goes well. 
Icon appears in the menu bar
But the errors above appear

The executable is in the Program folder not in one of the directories listed in the PATH environment. I think this could be the root or the problem.

actuallymentor, 2023-02-13:
Please follow the issue template so I can properly debug :)

closed 2023-02-13 by actuallymentor

henning-henemm, 2023-02-14:
I tried so. 

I fixed it by manually installing the CLI. It seems like the DMG didn’t install the command line executable in /usr/bin

Best
Henning