actuallymentor/battery #65: Does not work after wakeup from sleep

dalisoft, 2023-02-13:
**What is the issue? (required)**
Macbook still charging after wake-up even i'm set `battery maintain 75`

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Install `battery` via Homebrew (cask)
2. Configure and use as is
3. Set Macbook to sleep at night (or when you not work)
4. Wake-up then see
5. IT IS CHARGING AGAIN

**Expected behavior (required)**
Battery should continue discharge

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem. On a Mac you can click `control`+`command`+`shift`+`4` to create a screenshot that you can paste into this field with `command`+`4`.

**Error logs**
<details>

<summary>Logs</summary>

```
👾 Battery CLI logs:

02/13/23-11:34:47 - Battery at 97% (target 75%)
02/13/23-11:35:47 - Battery at 97% (target 75%)
02/13/23-11:36:47 - Battery at 97% (target 75%)
02/13/23-11:37:47 - Battery at 97% (target 75%)
02/13/23-11:38:47 - Battery at 97% (target 75%)
02/13/23-11:39:47 - Battery at 98% (target 75%)
02/13/23-11:40:47 - Battery at 98% (target 75%)
02/13/23-11:41:47 - Battery at 98% (target 75%)
02/13/23-11:42:47 - Battery at 98% (target 75%)
02/13/23-11:43:48 - Battery at 98% (target 75%)
02/13/23-11:44:48 - Battery at 98% (target 75%)
02/13/23-11:45:48 - Battery at 98% (target 75%)
02/13/23-11:46:48 - Battery at 98% (target 75%)
02/13/23-11:47:48 - Battery at 98% (target 75%)
02/13/23-11:48:48 - Battery at 100% (target 75%)
02/13/23-11:49:48 - Battery at 100% (target 75%)
02/13/23-11:50:48 - Battery at 100% (target 75%)
02/13/23-11:51:48 - Battery at 100% (target 75%)
02/13/23-11:52:48 - Battery at 100% (target 75%)
02/13/23-11:53:48 - Battery at 100% (target 75%)
02/13/23-11:54:48 - Battery at 100% (target 75%)
02/13/23-11:55:48 - Battery at 100% (target 75%)
02/13/23-11:56:48 - Battery at 100% (target 75%)
02/13/23-11:57:48 - Battery at 100% (target 75%)
02/13/23-11:58:48 - Battery at 100% (target 75%)
02/13/23-11:59:48 - Battery at 100% (target 75%)
02/13/23-12:00:48 - Battery at 100% (target 75%)
02/13/23-12:01:48 - Battery at 100% (target 75%)
02/13/23-12:02:49 - Battery at 100% (target 75%)
02/13/23-12:03:49 - Battery at 100% (target 75%)
02/13/23-12:04:49 - Battery at 100% (target 75%)
02/13/23-12:05:49 - Battery at 100% (target 75%)
02/13/23-12:06:49 - Battery at 100% (target 75%)
02/13/23-12:07:49 - Battery at 100% (target 75%)
02/13/23-12:08:49 - Battery at 100% (target 75%)
02/13/23-12:09:49 - Battery at 100% (target 75%)
02/13/23-12:10:49 - Battery at 100% (target 75%)
02/13/23-12:11:49 - Battery at 100% (target 75%)
02/13/23-12:12:49 - Battery at 100% (target 75%)
02/13/23-12:13:49 - Battery at 100% (target 75%)
02/13/23-12:14:49 - Battery at 100% (target 75%)
02/13/23-12:15:49 - Battery at 100% (target 75%)
02/13/23-12:16:49 - Battery at 100% (target 75%)
02/13/23-12:17:49 - Battery at 100% (target 75%)
02/13/23-12:18:49 - Battery at 100% (target 75%)
[... 157 lines cut ...]
Setting interface refresh speed to 10 minutes

📁 Config folder details:

total 152
drwxr-xr-x   6 dalisoft  staff   192B Feb 13 05:24 .
drwx-wx---+ 93 dalisoft  staff   2.9K Feb 13 05:24 ..
-rwxr-xr-x   1 dalisoft  staff    35K Feb 13 13:26 battery.log
-rwxr-xr-x   1 dalisoft  staff     5B Feb 13 13:23 battery.pid
-rw-r--r--   1 dalisoft  staff    28K Feb 13 13:26 gui.log
-rw-r--r--   1 dalisoft  staff     3B Feb 13 13:23 maintain.percentage

⚙️  Battery data:

02/13/23-13:26:50 - Battery at 100% (8:55 remaining), smc charging enabled
02/13/23-13:26:50 - Your battery is currently being maintained at 75%
Battery CLI utility v1.0.1

```

</details>

**Additional context (optional)**
Machine: Macbook M1
OS: macOS Monterey 12.6.3

actuallymentor, 2023-02-14:
Is your device in clamshell mode?

dalisoft, 2023-02-14:
@actuallymentor No, not in clamshell mode. Use it as intended. Just plugged in external display and sometimes lost charging due of lighting issues

actuallymentor, 2023-02-14:
Could you run `battery logs 10000` and post them? In your logs I can't see back to when the battery was 80

dalisoft, 2023-02-14:
@actuallymentor I actually set my limit to 75% because that's +5% gives temperature +3-5C which is bad for battery (as tested myself)

<details>

<summary>Logs</summary>

```
👾 Battery CLI logs:

02/13/23-05:24:31 - Triggering discharge to 80 before enabling charging limiter
02/13/23-05:24:31 - Discharging to 80% from 100%
02/13/23-05:24:31 - 🔽🪫 Enabling battery discharging
02/13/23-05:24:31 - Battery at 100% (target 80%)
02/13/23-05:24:57 - Triggering discharge to 75 before enabling charging limiter
02/13/23-05:24:57 - Discharging to 75% from 100%
02/13/23-05:24:57 - 🔽🪫 Enabling battery discharging
02/13/23-05:24:57 - Battery at 100% (target 75%)
02/13/23-05:25:31 - Battery at 100% (target 80%)
02/13/23-05:25:57 - Battery at 100% (target 75%)
02/13/23-05:26:31 - Battery at 100% (target 80%)
02/13/23-05:26:57 - Battery at 100% (target 75%)
02/13/23-05:27:31 - Battery at 100% (target 80%)
02/13/23-05:27:57 - Battery at 100% (target 75%)
02/13/23-05:28:31 - Battery at 100% (target 80%)
02/13/23-05:28:57 - Battery at 100% (target 75%)
02/13/23-05:29:31 - Battery at 100% (target 80%)
02/13/23-05:29:57 - Battery at 100% (target 75%)
02/13/23-05:30:31 - Battery at 100% (target 80%)
02/13/23-05:30:57 - Battery at 100% (target 75%)
02/13/23-05:31:31 - Battery at 100% (target 80%)
02/13/23-05:31:57 - Battery at 100% (target 75%)
02/13/23-05:32:31 - Battery at 100% (target 80%)
02/13/23-05:32:57 - Battery at 100% (target 75%)
02/13/23-05:33:31 - Battery at 100% (target 80%)
02/13/23-05:33:57 - Battery at 100% (target 75%)
02/13/23-05:34:31 - Battery at 100% (target 80%)
02/13/23-05:34:57 - Battery at 100% (target 75%)
02/13/23-05:35:31 - Battery at 100% (target 80%)
02/13/23-05:35:58 - Battery at 100% (target 75%)
02/13/23-05:36:32 - Battery at 100% (target 80%)
02/13/23-05:36:58 - Battery at 100% (target 75%)
02/13/23-05:37:32 - Battery at 100% (target 80%)
02/13/23-05:37:58 - Battery at 100% (target 75%)
02/13/23-05:38:32 - Battery at 100% (target 80%)
02/13/23-05:38:58 - Battery at 100% (target 75%)
02/13/23-05:39:32 - Battery at 100% (target 80%)
02/13/23-05:39:58 - Battery at 100% (target 75%)
02/13/23-05:40:32 - Battery at 100% (target 80%)
02/13/23-05:40:58 - Battery at 100% (target 75%)
02/13/23-05:41:32 - Battery at 100% (target 80%)
02/13/23-05:41:58 - Battery at 100% (target 75%)
02/13/23-05:42:32 - Battery at 100% (target 80%)
02/13/23-05:42:58 - Battery at 100% (target 75%)
02/13/23-05:43:32 - Battery at 100% (target 80%)
02/13/23-05:43:58 - Battery at 100% (target 75%)
02/13/23-05:44:32 - Battery at 100% (target 80%)
02/13/23-05:44:58 - Battery at 100% (target 75%)
02/13/23-05:45:32 - Battery at 100% (target 80%)
02/13/23-05:45:58 - Battery at 100% (target 75%)
02/13/23-05:46:32 - Battery at 100% (target 80%)
02/13/23-05:46:58 - Battery at 100% (target 75%)
02/13/23-05:47:32 - Battery at 100% (target 80%)
02/13/23-05:47:58 - Battery at 100% (target 75%)
02/13/23-05:48:32 - Battery at 100% (target 80%)
02/13/23-05:48:58 - Battery at 100% (target 75%)
02/13/23-05:49:32 - Battery at 100% (target 80%)
02/13/23-05:49:58 - Battery at 100% (target 75%)
02/13/23-05:50:32 - Battery at 100% (target 80%)
02/13/23-05:50:58 - Battery at 100% (target 75%)
02/13/23-05:51:32 - Battery at 100% (target 80%)
[... 3169 lines cut ...]
Refreshing interface update timer
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Battery status: 
{"percentage":"75","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"75","battery_state":"75% (unknown remaining)","daemon_state":"smc charging disabled"}
Setting interface refresh speed to 10 minutes

📁 Config folder details:

total 512
drwxr-xr-x   6 dalisoft  staff   192B Feb 13 05:24 .
drwx-wx---+ 93 dalisoft  staff   2.9K Feb 14 17:56 ..
-rwxr-xr-x   1 dalisoft  staff    50K Feb 14 22:23 battery.log
-rwxr-xr-x   1 dalisoft  staff     5B Feb 13 13:23 battery.pid
-rw-r--r--   1 dalisoft  staff   133K Feb 14 22:19 gui.log
-rw-r--r--   1 dalisoft  staff     3B Feb 13 13:23 maintain.percentage

⚙️  Battery data:

02/14/23-22:23:34 - Battery at 75% (attached; remaining), smc charging disabled
02/14/23-22:23:34 - Your battery is currently being maintained at 75%
Battery CLI utility v1.0.1

```

</details>

sunknudsen, 2023-02-19:
Experiencing same issue… battery is discharging, then I put computer to sleep… when I wake computer the next morning, battery is at 100%.

actuallymentor, 2023-03-13:
Clamshell users, please try [this beta release](https://github.com/actuallymentor/battery/releases/tag/v1.1.4-beta) and thumbs up/down it to let me know if it works for you now!

There is now a new GUI option:

[image]

actuallymentor, 2023-03-14:
Closed by release [v1.1.4](https://github.com/actuallymentor/battery/releases/tag/v1.1.4). Please reopen if you have the same issue with that release (with debug info).

closed 2023-03-14 by actuallymentor