actuallymentor/battery #66: uninstallment guide

TheGlorySaint, 2023-02-13:
**What is the issue? (required)**
I'd like to see in the ReadMe a Guide how to completely uninstall battery and all its component and also the command line tool. What also could be possible, an uninstallation function inside battery itself.

**Expected behavior (required)**
A guide or uninstall button that uninstalls battery and its components.

actuallymentor, 2023-02-14:
`brew uninstall battery` should uninstall also components if you installed through brew. Please re-open if you find any lingering app files so I can investigate.

TheGlorySaint, 2023-02-14:
When I uninstalled battery with `brew uninstall battery` I still could use the battery command in my Terminal. After running `battery uninstall` when I wasn't able to run that command anymore.

actuallymentor, 2023-03-14:
There is now a GUI uninstall button, as well as a more complete command line uninstall. Thanks for reporting!

closed 2023-03-14 by actuallymentor