actuallymentor/battery #67: Install not working - Version 1.08

martin-77, 2023-02-14:
**What is the issue? (required)**
Install not working or not starting correctly 

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. install with brew install 
2. Following Error appear:

Battery limiter error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery maintain 80
/bin/bash: battery: command not found

Battery limiter error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

**Expected behavior (required)**
working install

**Error logs**
Battery limiter error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery maintain 80
/bin/bash: battery: command not found

Battery limiter error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

No idea how to solve this.
Macbook Pro M1 Max - Ventura 13.2

hermannrh88, 2023-02-14:
Same issue.

vertumnuss, 2023-02-14:
Same here MB Pro M1 Max - Ventura 13.2.1

Neifab, 2023-02-14:
Hello, same here, MBP M2 Max, Ventura 13.2.1

actuallymentor, 2023-02-14:
@martin-77 please use the issue template so I have debugging info.

closed 2023-02-14 by actuallymentor

martin-77, 2023-02-16:
I think i was using the Template. 
Pleased tell me, what exactly you do you need?