actuallymentor/battery #69: Icon Black or absent

jakjakob, 2023-02-14:
**What is the issue? (required)**
The icon in the icon bar is absent in dark mode (it was for a bit black, so not 100% sure), and black during light mode

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
I've done nothing

**Expected behavior (required)**
I should have a visible white icon
**Screenshots (optional)**
[image]
**Error logs**
👾 Battery CLI logs:

02/13/23-19:15:37 - Battery at 87% (target 80%)
02/13/23-19:16:37 - Battery at 87% (target 80%)
02/13/23-19:17:37 - Battery at 87% (target 80%)
02/13/23-19:18:37 - Battery at 87% (target 80%)
02/13/23-19:21:43 - Battery at 87% (target 80%)
02/13/23-19:22:43 - Battery at 87% (target 80%)
02/13/23-19:23:43 - Battery at 86% (target 80%)
02/13/23-19:24:43 - Battery at 86% (target 80%)
02/13/23-19:25:43 - Battery at 86% (target 80%)
02/13/23-19:58:36 - Battery at 86% (target 80%)
02/13/23-19:59:36 - Battery at 86% (target 80%)
02/13/23-20:00:37 - Battery at 86% (target 80%)
02/13/23-20:01:37 - Battery at 86% (target 80%)
02/13/23-20:02:37 - Battery at 86% (target 80%)
02/13/23-20:03:37 - Battery at 86% (target 80%)
02/13/23-20:04:37 - Battery at 86% (target 80%)
02/13/23-20:05:37 - Battery at 86% (target 80%)
02/13/23-20:06:37 - Battery at 86% (target 80%)
02/13/23-20:07:37 - Battery at 86% (target 80%)
02/13/23-20:08:37 - Battery at 86% (target 80%)
02/13/23-20:09:37 - Battery at 86% (target 80%)
02/13/23-20:10:37 - Battery at 86% (target 80%)
02/13/23-20:11:37 - Battery at 86% (target 80%)
02/13/23-20:12:37 - Battery at 86% (target 80%)
02/13/23-20:13:37 - Battery at 86% (target 80%)
02/13/23-20:14:37 - Battery at 86% (target 80%)
02/13/23-20:15:37 - Battery at 86% (target 80%)
02/13/23-20:16:37 - Battery at 86% (target 80%)
02/13/23-20:17:37 - Battery at 86% (target 80%)
02/13/23-20:18:37 - Battery at 86% (target 80%)
02/13/23-20:19:37 - Battery at 86% (target 80%)
02/13/23-20:20:37 - Battery at 86% (target 80%)
02/13/23-20:21:37 - Battery at 86% (target 80%)
02/13/23-20:22:38 - Battery at 86% (target 80%)
02/13/23-20:23:38 - Battery at 86% (target 80%)
02/13/23-20:24:38 - Battery at 86% (target 80%)
02/13/23-20:25:38 - Battery at 86% (target 80%)
02/13/23-20:26:38 - Battery at 86% (target 80%)
02/13/23-20:27:38 - Battery at 86% (target 80%)
02/13/23-20:28:38 - Battery at 86% (target 80%)
02/13/23-20:29:38 - Battery at 86% (target 80%)
02/13/23-20:30:38 - Battery at 86% (target 80%)
02/13/23-20:31:38 - Battery at 86% (target 80%)
02/13/23-20:32:38 - Battery at 85% (target 80%)
02/13/23-20:33:38 - Battery at 85% (target 80%)
02/13/23-20:34:38 - Battery at 85% (target 80%)
02/13/23-20:35:38 - Battery at 85% (target 80%)
02/13/23-20:36:38 - Battery at 85% (target 80%)
02/13/23-20:37:38 - Battery at 85% (target 80%)
02/13/23-20:38:38 - Battery at 85% (target 80%)
02/13/23-20:39:38 - Battery at 85% (target 80%)
02/13/23-20:40:38 - Battery at 85% (target 80%)
02/13/23-20:41:38 - Battery at 85% (target 80%)
02/13/23-20:42:38 - Battery at 84% (target 80%)
02/13/23-20:43:39 - Battery at 84% (target 80%)
02/13/23-20:44:39 - Battery at 84% (target 80%)
[... 137 lines cut ...]
Refreshing interface update timer
Display percentage undefined based on 18
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Resetting interface timer speed
Get active logo for 18
Generate app menu percentage: 18
Battery status: 
{"percentage":"18","remaining":"2:50","charging":true,"discharging":false,"maintain_percentage":"80","battery_state":"18% (2:50 remaining)","daemon_state":"smc charging enabled"}
Setting interface refresh speed to 10 minutes

📁 Config folder details:

total 432
drwxr-xr-x@  6 jakjakob  staff   192B 13 Feb 16:15 .
drwxr-x---+ 24 jakjakob  staff   768B 14 Feb 10:45 ..
-rwxr-xr-x@  1 jakjakob  staff    11K 14 Feb 21:08 battery.log
-rwxr-xr-x@  1 jakjakob  staff     6B 13 Feb 16:15 battery.pid
-rw-r--r--@  1 jakjakob  staff   141K 14 Feb 21:08 gui.log
-rw-r--r--@  1 jakjakob  staff     3B 13 Feb 16:15 maintain.percentage

⚙️  Battery data:

02/14/23-21:08:57 - Battery at 18% (2:50 remaining), smc charging enabled
02/14/23-21:08:57 - Your battery is currently being maintained at 80%
Battery CLI utility v1.0.1

jakjakob, 2023-02-14:
Update: I charged it and then it went back normal (White)

Here the log if needed

👾 Battery CLI logs:

02/13/23-19:17:37 - Battery at 87% (target 80%)
02/13/23-19:18:37 - Battery at 87% (target 80%)
02/13/23-19:21:43 - Battery at 87% (target 80%)
02/13/23-19:22:43 - Battery at 87% (target 80%)
02/13/23-19:23:43 - Battery at 86% (target 80%)
02/13/23-19:24:43 - Battery at 86% (target 80%)
02/13/23-19:25:43 - Battery at 86% (target 80%)
02/13/23-19:58:36 - Battery at 86% (target 80%)
02/13/23-19:59:36 - Battery at 86% (target 80%)
02/13/23-20:00:37 - Battery at 86% (target 80%)
02/13/23-20:01:37 - Battery at 86% (target 80%)
02/13/23-20:02:37 - Battery at 86% (target 80%)
02/13/23-20:03:37 - Battery at 86% (target 80%)
02/13/23-20:04:37 - Battery at 86% (target 80%)
02/13/23-20:05:37 - Battery at 86% (target 80%)
02/13/23-20:06:37 - Battery at 86% (target 80%)
02/13/23-20:07:37 - Battery at 86% (target 80%)
02/13/23-20:08:37 - Battery at 86% (target 80%)
02/13/23-20:09:37 - Battery at 86% (target 80%)
02/13/23-20:10:37 - Battery at 86% (target 80%)
02/13/23-20:11:37 - Battery at 86% (target 80%)
02/13/23-20:12:37 - Battery at 86% (target 80%)
02/13/23-20:13:37 - Battery at 86% (target 80%)
02/13/23-20:14:37 - Battery at 86% (target 80%)
02/13/23-20:15:37 - Battery at 86% (target 80%)
02/13/23-20:16:37 - Battery at 86% (target 80%)
02/13/23-20:17:37 - Battery at 86% (target 80%)
02/13/23-20:18:37 - Battery at 86% (target 80%)
02/13/23-20:19:37 - Battery at 86% (target 80%)
02/13/23-20:20:37 - Battery at 86% (target 80%)
02/13/23-20:21:37 - Battery at 86% (target 80%)
02/13/23-20:22:38 - Battery at 86% (target 80%)
02/13/23-20:23:38 - Battery at 86% (target 80%)
02/13/23-20:24:38 - Battery at 86% (target 80%)
02/13/23-20:25:38 - Battery at 86% (target 80%)
02/13/23-20:26:38 - Battery at 86% (target 80%)
02/13/23-20:27:38 - Battery at 86% (target 80%)
02/13/23-20:28:38 - Battery at 86% (target 80%)
02/13/23-20:29:38 - Battery at 86% (target 80%)
02/13/23-20:30:38 - Battery at 86% (target 80%)
02/13/23-20:31:38 - Battery at 86% (target 80%)
02/13/23-20:32:38 - Battery at 85% (target 80%)
02/13/23-20:33:38 - Battery at 85% (target 80%)
02/13/23-20:34:38 - Battery at 85% (target 80%)
02/13/23-20:35:38 - Battery at 85% (target 80%)
02/13/23-20:36:38 - Battery at 85% (target 80%)
02/13/23-20:37:38 - Battery at 85% (target 80%)
02/13/23-20:38:38 - Battery at 85% (target 80%)
02/13/23-20:39:38 - Battery at 85% (target 80%)
02/13/23-20:40:38 - Battery at 85% (target 80%)
02/13/23-20:41:38 - Battery at 85% (target 80%)
02/13/23-20:42:38 - Battery at 84% (target 80%)
02/13/23-20:43:39 - Battery at 84% (target 80%)
02/13/23-20:44:39 - Battery at 84% (target 80%)
02/13/23-20:45:39 - Battery at 84% (target 80%)
02/13/23-20:46:39 - Battery at 84% (target 80%)
02/13/23-20:47:39 - Battery at 84% (target 80%)
02/13/23-20:48:39 - Battery at 84% (target 80%)
02/13/23-20:49:39 - Battery at 84% (target 80%)
02/13/23-20:50:39 - Battery at 84% (target 80%)
02/13/23-20:51:39 - Battery at 84% (target 80%)
02/13/23-20:52:39 - Battery at 83% (target 80%)
02/13/23-20:53:39 - Battery at 83% (target 80%)
02/13/23-20:54:39 - Battery at 83% (target 80%)
[... 130 lines cut ...]
Resetting interface timer speed
Display percentage 80 based on 80
Generate app menu percentage: 80
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Refreshing interface update timer
Get active logo for 80
Setting interface refresh speed to 10 minutes
Battery status: 
{"percentage":"80","remaining":"unknown","charging":false,"discharging":false,"maintain_percentage":"80","battery_state":"80% (unknown remaining)","daemon_state":"smc charging disabled"}

📁 Config folder details:

total 560
drwxr-xr-x@  6 jakjakob  staff   192B 13 Feb 16:15 .
drwxr-x---+ 24 jakjakob  staff   768B 14 Feb 10:45 ..
-rwxr-xr-x@  1 jakjakob  staff    11K 14 Feb 23:08 battery.log
-rwxr-xr-x@  1 jakjakob  staff     6B 13 Feb 16:15 battery.pid
-rw-r--r--@  1 jakjakob  staff   237K 14 Feb 23:07 gui.log
-rw-r--r--@  1 jakjakob  staff     3B 13 Feb 16:15 maintain.percentage

⚙️  Battery data:

02/14/23-23:08:16 - Battery at 80% (attached; remaining), smc charging disabled
02/14/23-23:08:16 - Your battery is currently being maintained at 80%
Battery CLI utility v1.0.1

jakjakob, 2023-02-15:
Also shows wrong symbol->it shows as it is 50% at 79%
[image]

👾 Battery CLI logs:

02/13/23-19:22:43 - Battery at 87% (target 80%)
02/13/23-19:23:43 - Battery at 86% (target 80%)
02/13/23-19:24:43 - Battery at 86% (target 80%)
02/13/23-19:25:43 - Battery at 86% (target 80%)
02/13/23-19:58:36 - Battery at 86% (target 80%)
02/13/23-19:59:36 - Battery at 86% (target 80%)
02/13/23-20:00:37 - Battery at 86% (target 80%)
02/13/23-20:01:37 - Battery at 86% (target 80%)
02/13/23-20:02:37 - Battery at 86% (target 80%)
02/13/23-20:03:37 - Battery at 86% (target 80%)
02/13/23-20:04:37 - Battery at 86% (target 80%)
02/13/23-20:05:37 - Battery at 86% (target 80%)
02/13/23-20:06:37 - Battery at 86% (target 80%)
02/13/23-20:07:37 - Battery at 86% (target 80%)
02/13/23-20:08:37 - Battery at 86% (target 80%)
02/13/23-20:09:37 - Battery at 86% (target 80%)
02/13/23-20:10:37 - Battery at 86% (target 80%)
02/13/23-20:11:37 - Battery at 86% (target 80%)
02/13/23-20:12:37 - Battery at 86% (target 80%)
02/13/23-20:13:37 - Battery at 86% (target 80%)
02/13/23-20:14:37 - Battery at 86% (target 80%)
02/13/23-20:15:37 - Battery at 86% (target 80%)
02/13/23-20:16:37 - Battery at 86% (target 80%)
02/13/23-20:17:37 - Battery at 86% (target 80%)
02/13/23-20:18:37 - Battery at 86% (target 80%)
02/13/23-20:19:37 - Battery at 86% (target 80%)
02/13/23-20:20:37 - Battery at 86% (target 80%)
02/13/23-20:21:37 - Battery at 86% (target 80%)
02/13/23-20:22:38 - Battery at 86% (target 80%)
02/13/23-20:23:38 - Battery at 86% (target 80%)
02/13/23-20:24:38 - Battery at 86% (target 80%)
02/13/23-20:25:38 - Battery at 86% (target 80%)
02/13/23-20:26:38 - Battery at 86% (target 80%)
02/13/23-20:27:38 - Battery at 86% (target 80%)
02/13/23-20:28:38 - Battery at 86% (target 80%)
02/13/23-20:29:38 - Battery at 86% (target 80%)
02/13/23-20:30:38 - Battery at 86% (target 80%)
02/13/23-20:31:38 - Battery at 86% (target 80%)
02/13/23-20:32:38 - Battery at 85% (target 80%)
02/13/23-20:33:38 - Battery at 85% (target 80%)
02/13/23-20:34:38 - Battery at 85% (target 80%)
02/13/23-20:35:38 - Battery at 85% (target 80%)
02/13/23-20:36:38 - Battery at 85% (target 80%)
02/13/23-20:37:38 - Battery at 85% (target 80%)
02/13/23-20:38:38 - Battery at 85% (target 80%)
02/13/23-20:39:38 - Battery at 85% (target 80%)
02/13/23-20:40:38 - Battery at 85% (target 80%)
02/13/23-20:41:38 - Battery at 85% (target 80%)
02/13/23-20:42:38 - Battery at 84% (target 80%)
02/13/23-20:43:39 - Battery at 84% (target 80%)
02/13/23-20:44:39 - Battery at 84% (target 80%)
02/13/23-20:45:39 - Battery at 84% (target 80%)
02/13/23-20:46:39 - Battery at 84% (target 80%)
02/13/23-20:47:39 - Battery at 84% (target 80%)
02/13/23-20:48:39 - Battery at 84% (target 80%)
02/13/23-20:49:39 - Battery at 84% (target 80%)
02/13/23-20:50:39 - Battery at 84% (target 80%)
02/13/23-20:51:39 - Battery at 84% (target 80%)
02/13/23-20:52:39 - Battery at 83% (target 80%)
02/13/23-20:53:39 - Battery at 83% (target 80%)
02/13/23-20:54:39 - Battery at 83% (target 80%)
02/13/23-20:55:39 - Battery at 83% (target 80%)
02/13/23-20:56:39 - Battery at 83% (target 80%)
02/13/23-20:57:39 - Battery at 83% (target 80%)
02/13/23-20:58:39 - Battery at 83% (target 80%)
[... 129 lines cut ...]
Refreshing interface update timer
Battery status: 
{"percentage":"74","remaining":"unknown","charging":true,"discharging":false,"maintain_percentage":"80","battery_state":"74% (unknown remaining)","daemon_state":"smc charging enabled"}
Setting interface refresh speed to 10 minutes
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Refreshing interface update timer
Battery status: 
{"percentage":"74","remaining":"unknown","charging":true,"discharging":false,"maintain_percentage":"80","battery_state":"74% (unknown remaining)","daemon_state":"smc charging enabled"}
Setting interface refresh speed to 10 minutes

📁 Config folder details:

total 656
drwxr-xr-x@  6 jakjakob  staff   192B 13 Feb 16:15 .
drwxr-x---+ 25 jakjakob  staff   800B 14 Feb 23:46 ..
-rwxr-xr-x@  1 jakjakob  staff    11K 15 Feb 11:54 battery.log
-rwxr-xr-x@  1 jakjakob  staff     6B 13 Feb 16:15 battery.pid
-rw-r--r--@  1 jakjakob  staff   276K 15 Feb 11:52 gui.log
-rw-r--r--@  1 jakjakob  staff     3B 13 Feb 16:15 maintain.percentage

⚙️  Battery data:

02/15/23-11:54:14 - Battery at 74% ((no remaining), smc charging enabled
02/15/23-11:54:14 - Your battery is currently being maintained at 80%
Battery CLI utility v1.0.1

Neifab, 2023-02-15:
Hello, same problem on MacOS Ventura 13.2.1

[image]

earthsound, 2023-02-15:
duplicate of #56 and/or #57?

jakjakob, 2023-02-16:
> duplicate of #56 and/or #57?

Yes, with a few problems more (wrong battery charging indicator, and once it was completely absent (but clicking in the place where it should be worked))

earthsound, 2023-02-16:
> > duplicate of #56 and/or #57?
> 
> Yes, but here I mentions also that the status of the battery is wrong

Not necessarily wrong, that's just the image used for any percentage from 50-79. See the assets in [this folder](https://github.com/actuallymentor/battery/tree/main/app/assets).

jakjakob, 2023-02-16:
> > > duplicate of #56 and/or #57?
> > 
> > 
> > Yes, but here I mentions also that the status of the battery is wrong
> 
> Not necessarily wrong, that's just the image used for any percentage from 50-79. See the assets in [this folder](https://github.com/actuallymentor/battery/tree/main/app/assets).

but still confusing... 79 is way closer to 80 than 50

closed 2023-03-06 by actuallymentor