actuallymentor/battery #70: does not work after installation

counterpointer, 2023-02-14:
I installed 1.0.8 with dmg.
Installation completed but after entering admin password I had 3 error messages.

> Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
> Error getting battery status: 
> Error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
> /bin/bash: battery: command not found
> 
> Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery maintain 80
> Error enabling battery: 
> Error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery maintain 80
> /bin/bash: battery: command not found
> 
> Refreshing tray icon...
> Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
> Error getting battery status: 
> Error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
> /bin/bash: battery: command not found

I can't quit the app nor uninstall it

Neifab, 2023-02-15:
Had the same, but no problems with Homebrew install.

counterpointer, 2023-02-15:
Yes, it works. Thanks!
I had not Homebrew installed, so it takes time.

closed 2023-02-15 by counterpointer

galaxydpw, 2023-02-17:
installed via homebrew, and same issue.

[image]