actuallymentor/battery #71: Make MagSafe charging LED green when 80%

jakjakob, 2023-02-14:
**Is your feature request related to a problem? Please describe.**
I want to have my battery at 80%, but since there is the possibility to check even if the computer is closed (the MagSafe LED), I don't want to miss it out using battery

**Describe the solution you'd like**
Whenever the battery reaches 80% (the charging limit) the MagSafe LED should turn green

**Describe alternatives you've considered**
Opening the MacBook (but impractical)

NdR91, 2023-02-22:
I'm not sure this is possible...

jakjakob, 2023-03-13:
It is! Al Dente did it.

omniwired, 2023-03-14:
not sure how they did it but here is more information about it.
https://apphousekitchen.com/feature-explanation-control-magsafe-led/

Is part of the paid version, so the code is not open source.

jonahclarsen, 2023-03-14:
This would be amazing! I wonder if anyone has open-sourced any code to do this. I personally would prefer to just disable the Magsafe LED altogether, at least while charged.

actuallymentor, 2023-03-15:
If you can find the `SMC` key for it I'm happy to add it.

jonahclarsen, 2023-03-16:
I looked more into it and I couldn't find the SMC key but it seems like the AlDente team may have taken the code from an open-source project, so it shouldn't be too hard to find. I asked about it on stackoverflow and the answer I got seems to point in the right direction: [https://stackoverflow.com/questions/75738532/control-magsafe-leds-on-apple-silicon-macbook/75738890#75738890](https://stackoverflow.com/questions/75738532/control-magsafe-leds-on-apple-silicon-macbook/75738890#75738890)

omniwired, 2023-03-16:
Nice find!
It seems that is different for M1 macs and intel macs

jonahclarsen, 2023-03-16:
@omniwired It seems that app is Intel only, but I don't see any indication that the SMC key for MagSafe LEDs is different on Apple Silicon macs, are you seeing that somewhere?

omniwired, 2023-03-17:
I have an M1 mac and use the key used in https://github.com/zackelia/bclm/blob/master/Sources/bclm/main.swift#L56

I've got this:
```
 /usr/local/bin/smc -k BFCL -r  
  BFCL  [    ]  no data
  ```
That makes me think it might be different.

@actuallymentor this is the key to use `BFCL`.

actuallymentor, 2023-07-20:
@omniwired ah thank you for finding it! Are you open to creating a PR for this?

ibrado, 2023-08-23:
> If you can find the `SMC` key for it I'm happy to add it.

From the [batt project](https://github.com/charlie0129/batt),

[ACLC](https://github.com/charlie0129/batt/blob/2a124d780f37035002a0dc0e3ed22132c87f68da/smc/consts.go#L5)
[03, 04](https://github.com/charlie0129/batt/blob/2a124d780f37035002a0dc0e3ed22132c87f68da/smc/magsafe.go#L11-L12)

```
        sudo smc -k ACLC -w 04 # orange
        sudo smc -k ACLC -w 03 # green
```

Tried the above; they work! 😄

jakjakob, 2023-08-24:
> From the [batt project](https://github.com/charlie0129/batt),
>
> [ACLC](https://github.com/charlie0129/batt/blob/2a124d780f37035002a0dc0e3ed22132c87f68da/smc/consts.go#L5) [03, [...]

obv. with sudo sec -k ACLC -w 

01 off
06 slow blinking orange
07 fast blinking orange
16 blinking
17 blinking

25 blinks off (blinks a few times and LED gets off)

It would be nice to have an option to turn it completely off, or keep it blinking while in battery maintain? (and maybe add a no-blink option bc it is prob. a bit distracting)

jakjakob, 2023-08-24:
I tries it until 25, may try other later on; for the people wondering the other codes just are repetitions from before (plain green, plain orange, off). The blinking are all different speed, that's why I mentioned them all

actuallymentor, 2023-08-24:
@jakjakob I might play with this today. Is there a code for "resume automatic behaviour"?

jakjakob, 2023-08-24:
> @jakjakob I might play with this today. Is there a code for "resume automatic behaviour"?

I don't know, since I haven't tested it yet with battery full loaded. Maybe try the first few codes and see if they change behavior with battery at 100% and not. It could be 02 or 00, since both gave orange output while having battery under 100%. I'll try also if maybe replugging the charger resets it. Do you know any way to find out which code it is currently running? Maybe so we can find out what the dafault(s) are

actuallymentor, 2023-08-24:
Allright,[ new release published!](https://github.com/actuallymentor/battery/releases/tag/v1.2.0)

Could you test it @jakjakob? I'm coding in a car without power.

omniwired, 2023-08-24:
I think you meant 71 on the release notes

actuallymentor, 2023-08-24:
@omniwired, yes, yes I did. Edited, thanks for pointing that out

tchajed, 2023-08-24:
This worked for me; after updating and restarting my MagSafe LED changed from orange to green. It seems like such a small thing but it feel great to have a green LED. Thanks for your work on this everyone!

actuallymentor, 2023-08-24:
Thanks for confirming @tchajed! I will close this issue. If anyone encounters an issue I will reopen it.

closed 2023-08-24 by actuallymentor

omniwired, 2023-08-25:
it works really well!

ibrado, 2023-08-26:
Needs a little improvement... if you connect the MagSafe charger when the battery is already at 80%+ , it will show an orange LED that doesn't eventually turn green. Menu correctly says "Power: smc charging disabled"

omniwired, 2023-08-28:
@ibrado @actuallymentor if you disable and then re-enable the limit, it changes to green

ibrado, 2023-09-19:
Submitted PR #184

bmarkkkkk, 2023-12-29:
hey guys! thanks for the great work. so this definitely does work, but one issues is if the battery is already at 80% or above, and you plug the charger, it will remain orange forever, would be great if you can poll the charge status and set the LED to green whenever it is 80% or greater, rather than crossing. thanks!