actuallymentor/battery #74: Battery limiter error

mwennrich81, 2023-02-15:
**What is the issue? (required)**

Battery limiter error: Command failed: PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
/bin/bash: battery: command not found

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:

Directly after the installation 

**Expected behavior (required)**

**Screenshots (optional)**

[image]

**Additional context (optional)**
Add any other context about the problem here.

PeterPung, 2023-05-07:
Got exactly same problem after installed 1.1.4

actuallymentor, 2023-07-20:
@PeterPung would you mind resubmitting this with the issue template (which I created only after your issue)? It has the debugging steps I need :)

closed 2023-07-20 by actuallymentor