actuallymentor/battery #75: Disable forcing discharge

eschrade98, 2023-02-15:
**Is your feature request related to a problem? Please describe.**
When I connect my Macbook to my docking station with more than 80% battery left, I can not close the lid anymore, as the MB discharges and goes to sleeping mode. With that, my external display and also my docking station turns off.

**Describe the solution you'd like**
A switch to decide wether to discharger from the battery down to 80%, or to stay at the current battery level, would be awesome.

**Describe alternatives you've considered**
Only alternative right now is to not close the lid until 80% is reached.

**Additional context**
Thank so much for your application, I like it really much :)

actuallymentor, 2023-02-19:
The issue only exists when you are discharging and close the screen?

I'm not planning to add the feature, but would welcome a pull-request for it :)

(Alternatively you can use the CLI if you comfortable in the terminal).

teegita, 2023-02-24:
I have the same issue using a Dell Monitor with integrated docking station. As soon as I enable the 80%-limit the monitor shuts off – currrently 100% battery ;-)
Also disconnecting the MacBook, opening the lid, reconnecting the dock and closing the lid again has the same result.

I assume if I go below 80% it shouldn’t happen. Right? Sorry for not testing before but I thought the issue description can be a bit more complete.

actuallymentor, 2023-02-28:
@teegita that is a different issue than described here.

teegita, 2023-02-28:
@actuallymentor Maybe I wrote my description wrong but my case fits exactly to @eschrade98’s comment:

> When I connect my Macbook to my docking station with more than 80% battery left, I can not close the lid anymore, as the MB discharges and goes to sleeping mode. With that, my external display and also my docking station turns off.

Now my battery is lower than 80%. I test later again to exclude other issues.

schabrolles, 2023-03-01:
This is the same issue than #59

actuallymentor, 2023-03-08:
Is a duplicate of #28 where we're tracking the issue. Closing this one to avoid duplicate discussions.

closed 2023-03-08 by actuallymentor