actuallymentor/battery #77: menu icon needs a negative glyph

winstonford, 2023-02-15:
The [sweet] menu bar icon shows black when others show white.  I believe what is needed is a single glyph mask, a negative, so that a white or black background will show thru glyph depending on dark or light mode in macOS.

[image]

winstonford, 2023-02-15:
Updated to 0.8, and issue persists. 

[image]

[image]

earthsound, 2023-02-15:
duplicate of #56 and/or #57?

actuallymentor, 2023-03-06:
Should be fixed in latest release [v1.1.1](https://github.com/actuallymentor/battery/releases/tag/v1.1.1)

closed 2023-03-06 by actuallymentor