actuallymentor/battery #79: Hi,

mumu38, 2023-02-16:
**What is the issue? (required)**
A clear and concise description of what the bug is.

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior (required)**
A clear and concise description of what you expected to happen.

**Screenshots (optional)**
If applicable, add screenshots to help explain your problem. On a Mac you can click `control`+`command`+`shift`+`4` to create a screenshot that you can paste into this field with `command`+`4`.

**Error logs**
If you know how a terminal works, please paste the content of `battery logs` here. This allows me to see potential errors on your device.

**Additional context (optional)**
Add any other context about the problem here.

mumu38, 2023-02-16:
Hi,
I tried to install the 1.0.8 version on a MacBook Pro M1 with Monterey.
I got the following errors during the update process after launching the app:

[image]
[image]
[image]

Is there something I missed? 

Thanks

earthsound, 2023-02-17:
duplicate of #74

mumu38, 2023-02-17:
OK. Just needed to install command tools.
Didn't even know those existed, even less installing them was needed.
Thanks

closed 2023-02-17 by mumu38