actuallymentor/battery #80: Issues with installation on new version with Brew and Bash

pkgprateek, 2023-02-17:
Steps to reproduce the behavior 2 ways:
1. Install `brew install battery`
2. Alfred -> battery
3. Error for screenshot

For Command Line method (Different Error):
1. Install using the given command in README
2. Check screenshopt for error/

[image]
[image]
[image]
[image]

galaxydpw, 2023-02-17:
on a Macbook Pro M2, 2023, and ventura 13.2.1
installed via homebrew:

[image]

hermannrh88, 2023-02-18:
Same issue MB Air M1 Ventura 13.2.1

actuallymentor, 2023-02-21:
Does this occur with the new `1.0.9` version too?

ballo, 2023-02-22:
seems broken:
> brew --prefix battery
Error: No available formula with the name "battery". Did you mean attr?

actuallymentor, 2023-02-24:
Could you guys try the new draft 1.0.10 release?

ballo, 2023-02-24:
Which one? brew or the app? brew rules seem broken

Aqua-Ye, 2023-02-27:
> Could you guys try the new draft 1.0.10 release?

In fact i'm not sure I have the same error as in this thread. It fails with this error: `xcode-select: error: command line tools are already installed, use "Software Update" in System Settings to install updates`

pkgprateek, 2023-02-27:
Same error latest version `1.0.10`

Tried using `.dmg`, `brew` and also manual `curl`

[image]

actuallymentor, 2023-03-07:
The new `1.1.x` versions do not use xcode anymore. Reopen this issue if it doesn't work for you.