actuallymentor/battery #81: "Battery limiter error: Cannot read properties..."

carlodrift, 2023-02-18:
I don't know what it is or what impact it has:
[image]

```
👾 Battery CLI logs:

02/15/23-10:14:23 - Battery at 86% (target 80%)
02/15/23-10:15:23 - Battery at 86% (target 80%)
02/15/23-10:16:23 - Battery at 86% (target 80%)
02/15/23-10:17:24 - Battery at 86% (target 80%)
02/15/23-10:18:24 - Battery at 86% (target 80%)
02/15/23-10:19:24 - Battery at 86% (target 80%)
02/15/23-10:20:24 - Battery at 86% (target 80%)
02/15/23-10:21:24 - Battery at 86% (target 80%)
02/15/23-10:22:24 - Battery at 85% (target 80%)
02/15/23-10:23:24 - Battery at 85% (target 80%)
02/15/23-10:24:24 - Battery at 85% (target 80%)
02/15/23-10:25:24 - Battery at 85% (target 80%)
02/15/23-10:26:24 - Battery at 84% (target 80%)
02/15/23-10:27:24 - Battery at 84% (target 80%)
02/15/23-10:28:24 - Battery at 84% (target 80%)
02/15/23-10:29:24 - Battery at 84% (target 80%)
02/15/23-10:30:24 - Battery at 84% (target 80%)
02/15/23-10:31:24 - Battery at 84% (target 80%)
02/15/23-10:32:24 - Battery at 83% (target 80%)
02/15/23-10:33:24 - Battery at 83% (target 80%)
02/15/23-10:34:24 - Battery at 83% (target 80%)
02/15/23-10:35:24 - Battery at 83% (target 80%)
02/15/23-10:36:25 - Battery at 83% (target 80%)
02/15/23-10:37:25 - Battery at 83% (target 80%)
02/15/23-10:38:25 - Battery at 82% (target 80%)
02/15/23-10:39:25 - Battery at 82% (target 80%)
02/15/23-10:40:25 - Battery at 82% (target 80%)
02/15/23-10:41:25 - Battery at 82% (target 80%)
02/15/23-10:42:25 - Battery at 82% (target 80%)
02/15/23-10:43:25 - Battery at 82% (target 80%)
02/15/23-10:44:25 - Battery at 81% (target 80%)
02/15/23-10:45:25 - Battery at 81% (target 80%)
02/15/23-10:46:25 - Battery at 81% (target 80%)
02/15/23-10:47:25 - Battery at 81% (target 80%)
02/15/23-10:48:25 - Battery at 81% (target 80%)
02/15/23-10:49:25 - Battery at 81% (target 80%)
02/15/23-10:50:25 - 🔼🪫 Disabling battery discharging
02/15/23-10:50:25 - Discharging completed at 80%
02/15/23-10:50:25 - Discharge pre battery-maintenance complete, continuing to battery maintenance loop
02/15/23-10:50:25 - Charging to and maintaining at 80% from 80%
02/15/23-10:50:25 - Charge above 80
02/15/23-10:50:25 - 🔌🪫 Disabling battery charging
02/15/23-17:07:20 - Debug trail. User: clement, config folder: /Users/clement/.battery, logfile: /Users/clement/.battery/battery.log, file called with 1: maintain_synchronous, 2: recover
02/15/23-17:07:20 - Recovering maintenance percentage 80
02/15/23-17:07:20 - Triggering discharge to 80 before enabling charging limiter
02/15/23-17:07:20 - Discharging to 80% from 80%
02/15/23-17:07:20 - 🔽🪫 Enabling battery discharging
02/15/23-17:07:24 - 🔼🪫 Disabling battery discharging
02/15/23-17:07:24 - Discharging completed at 80%
02/15/23-17:07:24 - Discharge pre battery-maintenance complete, continuing to battery maintenance loop
02/15/23-17:07:24 - Charging to and maintaining at 80% from 80%
02/15/23-17:07:24 - Charge above 80
02/15/23-17:07:24 - 🔌🪫 Disabling battery charging
02/16/23-05:33:59 - Charge below 80
02/16/23-05:33:59 - 🔌🔋 Enabling battery charging
02/16/23-05:33:59 - 🔼🪫 Disabling battery discharging
02/16/23-10:45:22 - Charge above 80
02/16/23-10:45:22 - 🔌🪫 Disabling battery charging
02/17/23-18:47:04 - Triggering discharge to 80 before enabling charging limiter
02/17/23-18:47:04 - Discharging to 80% from 80%
02/17/23-18:47:04 - 🔽🪫 Enabling battery discharging
02/17/23-18:47:04 - 🔼🪫 Disabling battery discharging
02/17/23-18:47:04 - Discharging completed at 80%
02/17/23-18:47:04 - Discharge pre battery-maintenance complete, continuing to battery maintenance loop
[... 130 lines cut ...]
Resetting interface timer speed
Executing PATH=$PATH:/bin:/usr/bin:/usr/local/bin:/usr/sbin:/opt/homebrew:/usr/bin/ battery status_csv
Generate app menu percentage: 77
Refreshing interface update timer
Get active logo for 77
Battery status: 
{"percentage":"77","remaining":"1:21","charging":true,"discharging":false,"maintain_percentage":"80","battery_state":"77% (1:21 remaining)","daemon_state":"smc charging enabled"}
Setting interface refresh speed to 10 minutes

📁 Config folder details:

total 784
drwxr-xr-x@   6 clement  staff   192B 18 fév 10:51 .
drwxr-xr-x+ 148 clement  staff   4,6K 18 fév 00:04 ..
-rwxr-xr-x@   1 clement  staff    14K 18 fév 10:51 battery.log
-rw-r--r--@   1 clement  staff     6B 18 fév 10:51 battery.pid
-rw-r--r--@   1 clement  staff   311K 18 fév 10:51 gui.log
-rw-r--r--@   1 clement  staff     3B 18 fév 10:51 maintain.percentage

⚙️  Battery data:

02/18/23-10:51:28 - Battery at 77% (1:21 remaining), smc charging enabled
02/18/23-10:51:28 - Your battery is currently being maintained at 80%
Battery CLI utility v1.0.1
```

actuallymentor, 2023-02-21:
I found the source of the issue and will include it in next release!

closed 2023-02-21 by actuallymentor

carlodrift, 2024-03-23:
No change, still the incessant popup in version 1.1.6