actuallymentor/battery #83: battery command stopped working

ballo, 2023-02-21:
**What is the issue? (required)**
After doing a brew upgrade, the command fails to work. It claims to be maintaining the battery level but isn't charging. Also, new error?
**What exactly did you do to produce the issue? (required)**
 > sudo battery maintain 40
01:16:11 - Starting battery maintenance at 40%
01:16:11 - Maintaining battery at 40%
(does not maintain at 40%)
 > sudo battery status
/usr/local/bin/battery: line 107: smc: command not found
01:16:22 - Battery at 27% (attached; remaining), smc charging disabled
01:16:22 - Your battery is currently being maintained at 40%
(smc command not found)
**Screenshots (optional)**

[image]

**Error logs**
> battery logs 
results in null output.

ballo, 2023-02-21:
M1 MaBook Pro 13"

actuallymentor, 2023-02-21:
You should never call `battery` with sudo. Please `battery uninstall` and reinstall.

Before reinstall please make sure you removed any artifacts that exist because you were using `sudo`:

```
sudo rm -rf ~/.battery
binfolder=/usr/local/bin
sudo rm -v "$binfolder/smc" "$binfolder/battery"
```

ballo, 2023-02-21:
OK, I uninstalled battery and the other stuff, and after a brew install battery:
> battery charge 40
zsh: command not found: battery

actuallymentor, 2023-02-24:
If you want the CLI: you need to open the GUI at least once or run the CLI-only script in the readme.

closed 2023-02-24 by actuallymentor

ballo, 2023-02-24:
Still broken 
[image]
[image]

ballo, 2023-02-24:
Why close this ticket when it's clearly still broken?