actuallymentor/battery #85: Can this app disable charging while MacBook is turned off?

isobeyasuaki, 2023-02-22:
**What is the issue? (required)**
Hello. I'm new to Mac.
I'm using MacBook Air (2022)(M2), with the minimum spec.
I'm charging my MacBook with a USB-C charger.
I enabled "battery" to stop charging at 80%, but my MacBook is being charged while it is switched off.
May I know if it is how it is supposed to work?
Or is there a way that I can stop charging while my MacBook is off?

OS: macOS Ventura version 13.0

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Go to "battery" GUI
2. Click on 'enable 80% battery limit'
3. Shut down PC
4. Battery is charged to 100% while PC is off

**Expected behavior (required)**
Battery charging stops at 80% even while PC is switched off

isobeyasuaki, 2023-02-22:
I'm using battery v1.0.9

actuallymentor, 2023-02-24:
I'm afraid when it is off, the program is unable to control anything :)

closed 2023-02-24 by actuallymentor