actuallymentor/battery #87: Problems when updating to version 1.0.9

Zeroscl, 2023-02-23:
When updating the version of Battery from 1.0.8, which is the one available by Brew, to 1.0.9 available by .DMG, I get the following error.

It asks me for authorization to install the xcode packages on my M1 MB Pro and remains in a continuous updating loop. 

[image]

[image]

Any idea how to solve? Has it happened to someone? It is assumed that I have that package installed since I program (study) in Python and for the latest version it is requested. 

Thanks in advance for the help.

Pablo

Zeroscl, 2023-02-23:
and to be able to get out of that loop I must use the OS activity monitor and kill the Battery process

actuallymentor, 2023-02-24:
Thanks for your report @Zeroscl!

Could you try [this new beta release](https://github.com/actuallymentor/battery/releases/tag/v1.0.10)?

And show me: `tail -n 1000 ~/.battery/*.log`?

Zeroscl, 2023-02-24:
@actuallymentor I am attaching the .log file you asked me for.

I just started my mac, and it sent me an update notice to version 1.0.9 (it updated itself) when indicating the restart that it asks for and trying to install the Xcode, it enters the infinite loop. Now I had to kill the process, and uninstall everything and try to downgrade to version 1.0.8 (the last one that worked well for me) but when I activate it immediately I get the 1.0.9 bug.

Do you recommend 1.0.10 even though it is in beta?

Thanks

[log file.pdf](https://github.com/actuallymentor/battery/files/10829143/log.file.pdf)

Zeroscl, 2023-02-24:
I am going to try to test the new version in Beta 1.0.10, @actuallymentor but in any case it would be interesting if there was the option to leave the auto update inactive.

Regards

actuallymentor, 2023-03-07:
@Zeroscl I'm closing this so we can continue in #98