actuallymentor/battery #89: Enabling charging limit puts the laptop to sleep when in clamshell mode.

cholcombe973, 2023-02-24:
**What is the issue? (required)**
I'm not entirely sure what is going on but it seems like when I enable the 80% charging limit while the laptop is in clamshell mode it goes to sleep. If I open the laptop up and log back in it goes right back to sleep. I'm currently on version 1.0.9

**What exactly did you do to produce the issue? (required)**
Steps to reproduce the behavior:
1. Install battery
2. Enable the 80% limit while in clamshell mode
3. Computer goes to sleep
4. Wake computer up and log in again
5. Disable 80% limit
6. Go back into clamshell mode
7. Enable 80% limit and computer goes back to sleep
8. See error logs

**Expected behavior (required)**
The behavior I'm looking for is allowing the charging limit to stay in place while in clamshell mode. 

**Error logs**
[battery_logs.gz](https://github.com/actuallymentor/battery/files/10826651/battery_logs.gz)

Did `battery` fail to install? Use `tail -n 1000 ~/.battery/*/log` instead.

**Additional context (optional)**

Add any other context about the problem here.

cholcombe973, 2023-02-24:
Maybe a potential solution for this is to have some code in place to detect when clamshell mode is on: https://unix.stackexchange.com/questions/11724/how-to-detect-a-laptop-hibernate-resumeclose-lid-open-in-osx-from-a-bash-scrip. It looks like a daemon can be registered for a callback to be hit.

The-Johnson, 2023-03-04:
This behavior is caused by the discharging function of the script. When your mac is above 80% battery will force it to use battery power in order to discharge down to <80%. Using battery power disables clamshell mode, due to apple's infinite wisdom that has determined that using clamshell on battery is wildly inappropriate. 

To fix this: go to /usr/local/bin/
open battery in your favorite text editor
Find these lines:
```
# Before we start maintaining the battery level, first discharge to the target level
log "Triggering discharge to $setting before enabling charging limiter"
battery discharge "$setting"
log "Discharge pre battery-maintenance complete, continuing to battery maintenance loop"
```

Add a # before battery discharge "$setting", like so:
`# battery discharge "$setting"`

This will disable the discharge behavior of battery, so you will have to manually get it down below 80%, but it will maintain the functionality of clamshell mode.

actuallymentor, 2023-03-13:
Clamshell users, please try [this beta release](https://github.com/actuallymentor/battery/releases/tag/v1.1.4-beta) and thumbs up/down it to let me know if it works for you now!

There is now a new GUI option:

[image]

actuallymentor, 2023-03-14:
Closed by release [v1.1.4](https://github.com/actuallymentor/battery/releases/tag/v1.1.4). Please reopen if you have the same issue with that release (with debug info).

closed 2023-03-14 by actuallymentor